microsoft/vscode-arduino #1428: Add status bar icon to send serial data

benmcmorran, 2022-01-20:
Thanks Ben. 

I added this code in the serialMonitor.js after line 54.
It gives an icon on the status bar for sending a serial message. I think it would be a great help to include it in future versions.
<details><summary> Code </summary>

in `serialMonitor.js` after line 54:

```
this._sendMessageStatusBar = vscode.window.createStatusBarItem(vscode.StatusBarAlignment.Right, constants.statusBarPriority.SEND_TEXT);
this._sendMessageStatusBar.command = "arduino.sendMessageToSerialPort";
this._sendMessageStatusBar.text = `$(comment-discussion)`;
this._sendMessageStatusBar.tooltip = "Send Message To Serial Port";
this._sendMessageStatusBar.show();
```

in `constants.js` after line 39:
```
`SEND_TEXT: 25,`
```
</details>

Now its not fool proof, it needs error checking to make sure the input box is not open. Not hard to implement. 
But yeah, I think this would really benefit the app. 

Best!

_Originally posted by @Triangle4 in https://github.com/microsoft/vscode-arduino/issues/1427#issuecomment-1016946730_

benmcmorran, 2022-01-20:
@gcampbell-msft FYI. Before making this change, we should evaluate if there are other places the icon could go, because this extension is already crowding the status bar.

Triangle4, 2022-01-20:
Hey Ben, 

So by quickly updating `updatePortStatus` in serialMonitor.js:

```
updatePortStatus(isOpened) {
        if (isOpened) {
            this._openPortStatusBar.command = "arduino.closeSerialMonitor";
            this._openPortStatusBar.text = `$(x)`;
            this._openPortStatusBar.tooltip = "Close Serial Monitor";
            this._baudRateStatusBar.show();
            this._sendMessageStatusBar.show();
        }
        else {
            this._openPortStatusBar.command = "arduino.openSerialMonitor";
            this._openPortStatusBar.text = `$(plug)`;
            this._openPortStatusBar.tooltip = "Open Serial Monitor (Read)";
            this._baudRateStatusBar.hide();
            this._sendMessageStatusBar.hide();
        }
}
```
and by removing `this._sendMessageStatusBar.show();` from the initial declaration, I hid the icon.
That free's up space as it is only there if the user really needs it.

Alternatively I created two svgs that matched (relatively) your teams styling, and follows their naming conventions:

<details><summary>ArduinoSendMessage_16.svg</summary>
```
<svg id="SVG" xmlns="http://www.w3.org/2000/svg" width="16" height="17.13" viewBox="0 0 16 17.13">
  <defs>
    <style>
      .cls-2 {
        fill: none;
        stroke: #fff;
        stroke-miterlimit: 10;
      }

      .cls-3 {
        fill: #231f20;
      }

      .cls-4 {
        font-size: 13.74px;
        fill: #00979c;
        font-family: ArialNarrow, Arial;
      }
    </style>
  </defs>
  <g id="BOX">
    <rect class="cls-2" x="2" y="2" width="12" height="12" rx="1"/>
  </g>
  <g id="CLI">
    <line class="cls-3" x1="8.29" y1="11.02" x2="12.29" y2="11.07"/>
    <line class="cls-2" x1="8.29" y1="11.02" x2="12.29" y2="11.07"/>
  </g>
  <text class="cls-4" transform="translate(3.77 12.92) scale(0.76 1)">&gt;</text>
</svg>
```
</details>

<details><summary>ArduinoSendMessage_inverse_16.svg</summary>
```
<svg id="SVG" xmlns="http://www.w3.org/2000/svg" width="16" height="17.13" viewBox="0 0 16 17.13">
  <defs>
    <style>
      .cls-1 {
        fill: #434343;
      }

      .cls-2 {
        fill: none;
        stroke: #fff;
        stroke-miterlimit: 10;
      }

      .cls-3 {
        fill: #231f20;
      }

      .cls-4 {
        font-size: 13.74px;
        fill: #00979c;
        font-family: ArialNarrow, Arial;
      }
    </style>
  </defs>
  <g id="BG">
    <rect class="cls-1" width="16" height="16" rx="2"/>
  </g>
  <g id="BOX">
    <rect class="cls-2" x="2" y="2" width="12" height="12" rx="1"/>
  </g>
  <g id="CLI">
    <line class="cls-3" x1="8.29" y1="11.02" x2="12.29" y2="11.07"/>
    <line class="cls-2" x1="8.29" y1="11.02" x2="12.29" y2="11.07"/>
  </g>
  <text class="cls-4" transform="translate(3.77 12.92) scale(0.76 1)">&gt;</text>
</svg>
```
</details>

and then added:
<details> <summary> to commands in package.json </summary>
{
        "command": "arduino.sendMessageToSerialPort",
        "title": "Arduino: Send Text to Serial Port",
        "icon": {
	        "dark": "images/ArduinoSendMessage_inverse_16.svg",
	        "light": "images/ArduinoSendMessage_16.svg"
        }
},

and :
<details> <summary> to menus in package.json </summary>
{
        {
	        "when": "resourceExtname == .ino",
	        "command": "arduino.sendMessageToSerialPort",
	        "group": "navigation"
        }
},

Hope that helps but its pretty basic stuff. I am sure no really needs my input! 
Feel free to reach out if you ever want help with anything :)

closed 2023-03-25 by github-actions[bot]