microsoft/vscode-arduino #1429: Allow baud rate to be set in arduino.json

tysonmatanich, 2022-01-21:
There is currently the ability to set `arduino.defaultBaudRate` in `settings.json` however no way to override it per project without manually changing it.

Please add support for a project level setting in `arduino.json`. For example:
```json
{
    "sketch": "example.ino",
    "board": "arduino:avr:uno",
    "port": "COM4",
    "baudRate": 9600,
    "output": "./build"
}
```

BelKed, 2022-01-21:
You can override it per each project using `settings.json` in `.vscode` folder (you have to create it in project directory). But you have to restart VSCode after each change of baud rate in that `.vscode/settings.json` file. This file should look like this:

```json
{
    "arduino.defaultBaudRate": 9600
}
```

---

But I think that, what you suggest is better, because after each change of baud rate using "Arduino: Change Baud Rate" it will be written in `arduino.json` and you don't have to restart VSCode after each change.

robotdad, 2022-02-04:
@benmcmorran what do you think of the suggestion for baud rate in arduino.json? Would arduino.json or settings.json be better in context of other changes going on around serial support? If we keep it in settings would it be possible to remove the requirement to reload VS to detect that change?

benmcmorran, 2022-02-04:
I'd prefer to keep settings in `settings.json` where possible. It plays better with the rest of the VS Code configuration ecosystem by automatically providing cascading overrides from the global to workspace folder level and has built-in graphical editor support. Regarding VS Code restarts after changing values in settings.json, we should be able to avoid that by subscribing to [`workspace.onDidChangeConfiguration`](https://code.visualstudio.com/api/references/vscode-api#workspace.onDidChangeConfiguration). I'll rename this issue to reflect that approach.

BelKed, 2022-02-04:
Also, after using command for changing value it should be written to `settings.json`.
And change of values in that file should reflect in VSCode...
 
By the way, isn't this issue related to this one?

- https://github.com/microsoft/vscode-arduino/issues/709

closed 2023-03-25 by github-actions[bot]