microsoft/vscode-arduino #1431: Serial port cannot open on version 0.4.10 on Windows 10

iqapps, 2022-01-24:
When clicking the open serial port icon, i get this error:

Failed to open serial port COM19 due to error: + TypeError: Cannot read property 'then' of undefined

Version 0.4.7 works with exact same setup without problems.

qweiAv, 2022-01-28:
I got the same error. However I tried it on a different PC and it works, which has a different VSCode version. Maybe try updating VSCode?

For reference: The VSCode version that gives this error is 1.60.0 (system setup). The one that doesn't have an error is 1.63.2 (user setup)

Also a side note: The PC that gives this error doesn't have admin privilege but the working PC does.

hamiltonwei, 2022-01-28:
I got the same error. However I tried it on a different PC and it works, which has a different VSCode version. Maybe try updating VSCode?

For reference: The VSCode version that gives this error is 1.60.0 (system setup). The one that doesn't have an error is 1.63.2 (user setup)

Also a side note: The PC that gives this error doesn't have admin privilege but the working PC does.

robotdad, 2022-02-07:
I'm not able to repro this either, I'm on v 1.64 of vscode. @iqapps can you provide additional repo steps or resolve if this is no longer an issue?

hamiltonwei, 2022-02-07:
I'm convinced that it's a problem with VS code 1.60.0. I upgraded to 1.64 and no longer has the issue.

benmcmorran, 2022-02-07:
@iqapps does updating VS Code itself also fix this issue for you?

iqapps, 2022-03-03:
Sorry for the long delay.
Yes, updating VSCode, now 1.64.* solved the issue.
I have had VSCode automatic updates set to off, due to problems with an other update earlier.

closed 2022-03-03 by iqapps