microsoft/vscode-arduino #1436: personal header files are not found if they are alphabetically first in the folder.

Triangle4, 2022-01-28:
Given an arduino file named:

```
m.ino
```

Now, should you have your own includes in the same directory, for example:
```
s.h
t.h
```

There will be no issue importing them.

```
#include "s.h"
#include "t.h"
```

However, should they be named as such:

```
#include "a.h"
#include "t.h"
```

I am recieving an error in this situation:

```
'a.h' No such file or directory.
```
renaming it to a number that alphabetically larger than the source file `m.ino` and deleting the c_cpp_properties.json and the output folder, results in a build that once again, works.

I am not sure if the is an arduino thing, or vscode thing, but its definitely a bug, at least for me.
I do not wish to use external library folders.

adiazulay, 2022-01-28:
I seem to remember a similar issue to this, and I think this happens if your parent directory name doesn't match the name of the .ino file.

benmcmorran, 2022-02-07:
I can reproduce this behavior, but only when the `.ino` file is named differently than the parent directory (e.g. `sketch.ino` in a directory named `myproject` doesn't work, but `myproject.ino` does). The requirement that the sketch and directory have the same name [comes from Arduino itself and is unlikely to change](https://github.com/arduino/Arduino/issues/10755#issuecomment-692036318). @Triangle4 does renaming your sketch or parent directory solve the issue?

Triangle4, 2022-02-07:
Thank you, I had no idea that requirement existed! I had a feeling it had to be something simple.

closed 2022-02-07 by Triangle4

benmcmorran, 2022-02-07:
Glad that works! I opened #1448 to track improving the warnings in the extension for this case.