microsoft/vscode-arduino #1437: arduino.path Windows 10 not recognized

Edodums, 2022-01-28:
I've compiled the input as it folllow "C:\Program Files (x86)\Arduino"

[image]

The path exists, I've used the Windows Store version until now, and I've download the normal version just for this extension.
Do I have to install in "Program Files" just to make it work?

[image]

Edodums, 2022-01-28:
Installing in "Program Files" doesn't change anything, for info I've the latest version of vscode.

Edodums, 2022-01-28:
"Note: Arduino IDE 1.8.7 had some breaking changes, causing board package and library installation failures. These failures were corrected in 1.8.8 and later."

Could be the problem linked to this?

Edodums, 2022-01-29:
It works only with Arduino CLI ( at least 😄 )

robotdad, 2022-02-07:
The current version of the Arduino IDE is 1.8.19 and this issue does not repro with it. The version you are citing, 1.8.7 is from 2018. Can you verify with a newer version or are you locked to that one?

Edodums, 2022-02-08:
I didn't explain my self well, I'm sorry for that.
I'm using the latest version of the arduino IDE ( I've got  two installations , one from downloadable installer , one from microsoft store )

davidcooper1, 2022-02-15:
Possibly related to #1451

benmcmorran, 2022-02-17:
Hi @Edodums! Can you provide the values of `arduino.path` and `arduino.commandPath` (if set) for your user and workspace settings? Additionally, can you verify that `arduino.exe`, `arduino-debug.exe`, and `arduino-builder.exe` are present in the directory specified by `arduino.path`?

Finally, any changes to `arduino.path` require a restart of VS Code before they take effect.

Edodums, 2022-02-17:
I'm sorry for not answered before but I'm a bit short on time in this period.
The path is the one that is written at the start of the issue.