microsoft/vscode-arduino #1438: Multiple .ino files and IntelliSense

BrandlMax, 2022-01-30:
Hi everyone,

In my Arduino projects, I like to divide my scripts into several files for overview and maintainability:

[image]

Unfortunately, IntelliSense has some trouble with these multiple files, as it does not recognize variables defined in the previous files and underlines them in red accordingly. 

This problem can be fixed if I write all .ino files manually into the c_cpp_properties.json:

[image]

**However, this file is automatically regenerated after each "verify" and "upload" and thus overwrites my changes in the "forcedInclude". So I then have to reinsert these lines each time.**

Since this question/problem has not been asked more often, I'm pretty sure I'm missing something fundamental. Could you help me? I would be very thankful to have this frustrating workflow a little more pleasant 😅

Thank you so much, 
Max

hlovdal, 2022-02-07:
Splitting up your project into multiple files based on functionality is indeed very good practice, however I think you will be better off by using `*.cpp` & `*.hpp` (or `*.c` & `*.h`) files rather than `*.ino`. The programming language used by Arduino is a variant of C++, and the ino files are sort of C++ files but with some [additional processing on top](https://arduino.github.io/arduino-cli/0.21/sketch-build-process/#pre-processing).

This is done to lower the barrier to getting started as low as possible and from that perspective it works, but it is not without problems (for instance having a function with an `enum` argument breaks this flow and needs to be manually corrected by injecting an explicit function prototype (one of the "In some rare cases" instances, which in my opinion should not be rare at all although maybe it is for beginners that use `int` for everything...)).

Multiple ino files will be merged into one file file during the build process (i.e. with no scope/namespace isolation), so using C or C++ files gives you better guards against writing spaghetti code where variables are modified all over the place (although if you want to you can expose variables from one file and make it available for modification globally you can do so with `extern`).

I have landed on having every single arduino project that I make have the exact same ino file

```C++
#include "src/main.hpp"

Main m;

void setup()
{
        m.setup();
}

void loop()
{
        m.loop();
}
```

where all the logic is put into a `Main` class with additional files for various functionality ([sort of the same advice](https://arduino.stackexchange.com/a/61636/3821)).

benmcmorran, 2022-02-07:
Thanks for the report @BrandlMax! I'm able to reproduce this issue by putting these two files in the same directory. The project verifies successfully, but IntelliSense squiggles the call to `test()`.

```c
sketch.ino
void setup() { test(); }
void looop() {}

test.ino
void test() {}
```

I'd echo the advice from @hlovdal above to consider using C/C++ directly instead of multiple `.ino` files. That said, this is definitely a bug in our IntelliSense implementation, so I'm leaving the issue open to track a fix.

BrandlMax, 2022-02-11:
Thank you, @hlovdal and @benmcmorran, 
And you both are right. Maybe it's really time to switch to C++ Files ;) 

I wish you both a great weekend!

palashbhowmick, 2022-07-24:
I'm also facing similar kind of issue. I have two `.ino` files - [one for Arduino UNO](https://github.com/palashbhowmick/robocar/blob/master/RoboCar.ino) and [another one for ESP8266 ESP-01](https://github.com/palashbhowmick/robocar/blob/master/RoboCarServer_ESP8266.ino). I want to keep two files in a single repository because they belongs to same project. 

Now, I can't set separate board configuration for these two files.

marcio-pessoa, 2023-02-21:
Hi dears.

The expected behavior using additional .ino files is described in the documentation:

> All .ino and .pde files in the sketch folder (shown in the Arduino IDE as tabs with no extension) are concatenated together, starting with the file that matches the folder name followed by the others in alphabetical order. The .cpp filename extension is then added to the resulting file.

Reference: [Sketch build process](https://arduino.github.io/arduino-cli/0.31/sketch-build-process/#pre-processing)

This is not the traditional behavior of C and C++ compilers, but Arduino's proposal is to bring practicality for the makers. So please, the concatenation of .ino files needs to be considered.

marcio-pessoa, 2023-04-13:
[image]

marcio-pessoa, 2023-04-13:
This issue is similar to #271 , perhaps this issue is duplicated.