microsoft/vscode-arduino #1442: does not build using recommended node version

austinoneil, 2022-02-04:
When building on OSX and running `gulp build`, it errors out on the webpack build stage.  This seems to be resolved when downgrading node to version 11.15.0.

benmcmorran, 2022-02-07:
Thanks for the report. What version of Node were you using before the downgrade? Our CI builds with [Node 14 on Mac without issues](https://github.com/microsoft/vscode-arduino/blob/08eabe6d845c024dd243b058f54a9f32b73e5296/.github/workflows/build.yml#L30-L37).

austinoneil, 2022-02-08:
Node is on v12.22.10.

npm is on 6.14.16.

I just tried it using Node v14.19.0 and npm v6.14.16 and I got the same error.

benmcmorran, 2022-02-09:
Hmm, okay. What's the error message? I made some [changes to the webpack configuration yesterday](https://github.com/microsoft/vscode-arduino/pull/1450/files#diff-1fb26bc12ac780c7ad7325730ed09fc4c2c3d757c276c3dacc44bfe20faf166f), but I don't see why that would be sensitive to Node version.

austinoneil, 2022-02-11:
Sorry for the late response. Was very busy yesterday.

[Here are the logs.](https://github.com/microsoft/vscode-arduino/files/8045270/gulp_logs.txt)

benmcmorran, 2022-02-11:
Thanks. Does it work if you run `npm rebuild node-sass --force` first? It looks like for some reason npm installed node-sass bindings for a different version of node. I'm also wondering if `npm install` ended up using a different version of Node than the Gulp tasks.

austinoneil, 2022-02-12:
It did not.  Is there a specific version of `node-sass` I need to be using?

BelKed, 2022-02-12:
I had the almost same problem with `node-sass`:
```
Found bindings for the following environments:
  - OS X 64-bit with Node.js 12.x
```

But after deleting all `node_modules` folders and fetching [recent changes from `package-lock.json`](https://github.com/microsoft/vscode-arduino/commit/fd921c387e4929150986c66de6940ec9fe965eaa#diff-053150b640a7ce75eff69d1a22cae7f0f94ad64ce9a855db544dda0929316519) it worked without problems.

davidcooper1, 2022-02-15:
@austinoneil Are you using nvm by any chance? I've noticed that, even if I set a node version using `nvm use`, the task runner will use the node version set as default in nvm.

austinoneil, 2022-02-15:
@davidcooper1 yes, I am using nvm.

benmcmorran, 2022-02-15:
Thanks all for the help! I'm renaming this issue to better reflect the discussion and will try to repro with nvm.

benmcmorran, 2022-02-17:
Reopening, #1461 likely does not fix this particular issue on Mac.

davidcooper1, 2022-02-24:
The previous fix for linux #1461 does still fix the path issue to find gulp, but I've noticed a weird recurrence of gulp using a different node version. Not sure if I accidentally installed gulp globally using node 17 then ran with node 14 though. I fixed it for now, but I'll keep an eye out.

closed 2022-04-29 by github-actions[bot]