microsoft/vscode-arduino #1445: SPIFFS Uploader

hdpklm, 2022-02-04:
Hello all

I use ESP32, windows 10

I need to upload file to SPIFFS from the vscode-arduino, not from the arduino IDE

i find the link for upload with arduino IDE
https://microcontrollerslab.com/install-esp32-filesystem-uploader-in-arduino-ide-spiffs/

thank you a lot.

benmcmorran, 2022-02-07:
Unfortunately, it looks like the SPIFFS uploader is a Java plugin for the Arduino IDE. vscode-arduino currently does not support these plugins and given that vscode-arduino only interacts with Arduino IDE on the command line, it would be difficult to add this support.

I'm marking this as a feature request and renaming the issue to better capture the problem. Although we don't have plans to fix this, I'm leaving the issue open in case others want to provide more information that would help us better prioritize this work.

robotdad, 2022-02-07:
Have you tried the Espressif IDF extension for VS Code? That can flash ESP32 boards.

hdpklm, 2022-02-10:
> Have you tried the Espressif IDF extension for VS Code? That can flash ESP32 boards.

yes but I dont know how to upload files in cli 🤷‍♂️