microsoft/vscode-arduino #1449: Intellisense can't find references, even in same file! Hover seems to work fine. VS Code, Arduino, SamD21

compulsivetinkerer, 2022-02-08:
I am having an odd problem getting Intellisense to "Find all references", "Go to definition", and "Go to references" within my workspace. It will never find references anywhere. "Go to definition" only works when the definition is in the same .ino file; if the definition is in any other file or library, it will go to the declaration in the appropriate header file. When I hover, I get valid "quick info", and when I start typing something, a valid completion list pops up, so intellisense is mostly working.

Now for the really odd part: I have recently moved from the Arduino IDE to VS Code using the Arduino Extension. When I did this, I got all the Intellisense features to work, even References. I then got my SEGGER debugger to work (a painfully trial-and-error process). When I tried References again, I found I had somehow broken it with the symptoms I described above.

Not wanting to mess up my debugger, I tried again to go through the whole process with my laptop (a similar machine to my desktop). In fairly short order, I got the laptop to work perfectly: Intellisense and debugger even References.

I then uninstalled VS Code on my desktop to start over from scratch, and I encountered the same References problems. As far as I can tell, both machines are set up the same. I even tried swapping c_cpp_properties.json files which made no difference.

My workspace is on a network drive, accessible from both machines as Y: drive. The single set of .json files are shared from this drive, but I experimented with directing both machines to different workspace foldere on that drive, and it made no difference.

Arduino 1.8.19

Board driver: SparkFun SAMD21 breakout dev card 1.8.6 (dependency: Arduino SAMD Boards 1.8.1)

Board driver: Arduino SAMD Boards 1.8.12

VS Code 1.64.0 Extensions:

Arduino microsoft version

Intellisense

C/C++

C/C++ Extension pack (9 Extensions)

Better C++ syntax

Remote Extensions (4)

`{
"version": 4,
"configurations": [
    {
        "name": "Arduino",
        "compilerPath": "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\bin\\arm-none-eabi-g++",
        "compilerArgs": [
            "-mcpu=cortex-m0plus",
            "-mthumb",
            "-w",
            "-std=gnu++11",
            "-ffunction-sections",
            "-fdata-sections",
            "-fno-threadsafe-statics",
            "-nostdlib",
            "--param",
            "-fno-rtti",
            "-fno-exceptions"
        ],
        "intelliSenseMode": "gcc-x64",
        "includePath": [
            "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0\\CMSIS\\Include\\",
            "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0\\CMSIS\\Device\\ATMEL\\",
            "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\cores\\arduino",
            "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\variants\\SparkFun_SAMD21_Dev",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\DougAS5600",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_BusIO",
            "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\libraries\\Wire",
            "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\libraries\\SPI",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_INA260_Library",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_AHRS\\src",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_Unified_Sensor",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_GPS_Library\\src",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Arduino-MemoryFree-master",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_SleepyDog_Library",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\WMM_Tinier\\src",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_LSM6DS",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_LIS3MDL",
            "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Iridium_SBD\\src",
            "C:\\Program Files (x86)\\Arduino\\libraries\\SD\\src",
[... 352 lines cut ...]
            "__LDBL_MIN_10_EXP__=(-307)",
            "__SIZEOF_LONG_LONG__=8",
            "__ULACCUM_EPSILON__=0x1P-32ULK",
            "__SACCUM_IBIT__=8",
            "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
            "__LDBL_DIG__=15",
            "__FLT_DECIMAL_DIG__=9",
            "__UINT_FAST16_MAX__=4294967295U",
            "__GNUC_GNU_INLINE__=1",
            "__GCC_ATOMIC_SHORT_LOCK_FREE=1",
            "__ULLFRACT_MAX__=0XFFFFFFFFFFFFFFFFP-64ULLR",
            "__UINT_FAST8_TYPE__=unsigned int",
            "__USFRACT_EPSILON__=0x1P-8UHR",
            "__ULACCUM_FBIT__=32",
            "__QQ_IBIT__=0",
            "__ATOMIC_ACQ_REL=4",
            "__ATOMIC_RELEASE=3",
            "USBCON"
        ]
    }
]`

Please help.  I bet it is something simple.

Doug

compulsivetinkerer, 2022-02-08:
My c_cpp_properties.json file was disappointing.  Let me try again:

`{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\bin\\arm-none-eabi-g++",
            "compilerArgs": [
                "-mcpu=cortex-m0plus",
                "-mthumb",
                "-w",
                "-std=gnu++11",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-nostdlib",
                "--param",
                "-fno-rtti",
                "-fno-exceptions"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0\\CMSIS\\Include\\",
                "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0\\CMSIS\\Device\\ATMEL\\",
                "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\cores\\arduino",
                "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\variants\\SparkFun_SAMD21_Dev",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\WMM_Tinier\\src",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_LSM6DS",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_BusIO",
                "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\libraries\\Wire",
                "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\libraries\\SPI",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_Unified_Sensor",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_LIS3MDL",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\DougAS5600",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_INA260_Library",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_AHRS\\src",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_GPS_Library\\src",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Iridium_SBD\\src",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Arduino-MemoryFree-master",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_SleepyDog_Library",
                "C:\\Program Files (x86)\\Arduino\\libraries\\SD\\src",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library",
                "C:\\Users\\doug\\Documents\\Arduino\\libraries\\Adafruit_SSD1306",
                "c:\\users\\doug\\appdata\\local\\arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\arm-none-eabi\\include\\c++\\4.8.3",
                "c:\\users\\doug\\appdata\\local\\arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\arm-none-eabi\\include\\c++\\4.8.3\\arm-none-eabi",
                "c:\\users\\doug\\appdata\\local\\arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\arm-none-eabi\\include\\c++\\4.8.3\\backward",
                "c:\\users\\doug\\appdata\\local\\arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\lib\\gcc\\arm-none-eabi\\4.8.3\\include",
                "c:\\users\\doug\\appdata\\local\\arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\lib\\gcc\\arm-none-eabi\\4.8.3\\include-fixed",
                "c:\\users\\doug\\appdata\\local\\arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\arm-none-eabi\\include"
            ],
            "forcedInclude": [
                "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\SparkFun\\hardware\\samd\\1.8.6\\cores\\arduino\\Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
                "F_CPU=48000000L",
                "ARDUINO=10819",
                "ARDUINO_SAMD_ZERO",
                "ARDUINO_ARCH_SAMD",
                "__SAMD21G18A__",
                "USB_VID=0x1B4F",
                "USB_PID=0x8D21",
                "USBCON",
                "USB_MANUFACTURER=\"SparkFun\"",
                "USB_PRODUCT=\"SFE SAMD21\"",
                "__DBL_MIN_EXP__=(-1021)",
                "__HQ_FBIT__=15",
                "__UINT_LEAST16_MAX__=65535",
[... 321 lines cut ...]
                "__DEC64_SUBNORMAL_MIN__=0.000000000000001E-383DD",
                "__ARM_EABI__=1",
                "__DEC128_MANT_DIG__=34",
                "__LDBL_MIN_10_EXP__=(-307)",
                "__SIZEOF_LONG_LONG__=8",
                "__ULACCUM_EPSILON__=0x1P-32ULK",
                "__SACCUM_IBIT__=8",
                "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
                "__LDBL_DIG__=15",
                "__FLT_DECIMAL_DIG__=9",
                "__UINT_FAST16_MAX__=4294967295U",
                "__GNUC_GNU_INLINE__=1",
                "__GCC_ATOMIC_SHORT_LOCK_FREE=1",
                "__ULLFRACT_MAX__=0XFFFFFFFFFFFFFFFFP-64ULLR",
                "__UINT_FAST8_TYPE__=unsigned int",
                "__USFRACT_EPSILON__=0x1P-8UHR",
                "__ULACCUM_FBIT__=32",
                "__QQ_IBIT__=0",
                "__ATOMIC_ACQ_REL=4",
                "__ATOMIC_RELEASE=3",
                "USBCON"
            ]
        }
    ]
}`

compulsivetinkerer, 2022-02-08:
I'm going to try one more time. 

 `
[c_cpp_properties .txt](https://github.com/microsoft/vscode-arduino/files/8026680/c_cpp_properties.txt)
`

benmcmorran, 2022-02-17:
Hi @compulsivetinkerer! IntelliSense is provided by the [C/C++ extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.cpptools) which should automatically be installed with the Arduino extension. `c_cpp_properties.json` should be the only file controlling the IntelliSense behavior, and by default it is automatically regenerated every time the Arduino project is verified or uploaded.

Can you provide the [C/C++ extension logs as described in this issue template](https://github.com/microsoft/vscode-cpptools/issues/new?assignees=&labels=&template=language-service.md&title=)? If it looks like the issue is caused by the C/C++ extension, I'll transfer this issue so that we can get the right team looking at it.

compulsivetinkerer, 2022-02-18:
- OS and Version:  **Windows 10** 
- VS Code Version:  **10.0.19043** 
- C/C++ Extension Version:  **v1.8.4**
- Other extensions you installed (and if the issue persists after disabling them):  **Arduino v0.4.11**
- If using SSH remote, specify OS of remote machine:  **v0.4.11**
- A clear and concise description of what the bug is, including information about the workspace (i.e. is the workspace a single project or multiple projects, size of the project, etc).
- 
-See the first post for a clear description------

**Steps to reproduce**

highlight variable **toggle** right click, click on "Find all references"  This is the one case where "Find all references works"

highlight constant BLUELED.  etc.  "Find all references" yields "no results",  Find Declaration does work.

highlight digitalWrite.  "Find all References" yields "no results"  "Find Definition"  yields the Declaration in awiring_digital.h, not the definition is a cpp file.

highlight flash.  "Find all references can't even find anything in the main file, let alone in function.ino

**Code sample and logs**  

- Code sample
------------- code compiles and runs just fine.  no problems, no squiggles.--------
You asked for this, but any code will demonstrate this.  There is nothing special about this code.
`#define BLUELED 13

void flash(uint16_t, bool);
 
void setup() {
pinMode(BLUELED, OUTPUT);
SerialUSB.println(" Starting Blue LED ");
}

void loop() {
    
  static bool toggle = 0;
  if(toggle = ! toggle){
    flash(BLUELED,LOW);
     SerialUSB.println(" Blue LED OFF ");
     delay(1000);
  }
  else{
    flash(BLUELED,HIGH);
     SerialUSB.println(" Blue LED ON ");
     delay(1000);
  }
}`
- Configurations in `c_cpp_properties.json`  
- Sorry about the formatting.  Tell me how, and I will edit this for you

`{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "C:\\Users\\doug\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1\\bin\\arm-none-eabi-g++",
            "compilerArgs": [
                "-mcpu=cortex-m0plus",
                "-mthumb",
                "-w",
                "-std=gnu++11",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-nostdlib",
                "--param",
                "-fno-rtti",
                "-fno-exceptions"
            ],
            "intelliSenseMode": "gcc-x64",
[... 1615 lines cut ...]
 * \note Digital pin PIN_LED is harder to use as a digital input than the other digital pins because it has an LED
 * and resistor attached to it that's soldered to the board on most boards. If you enable its internal 20k pull-up
 * resistor, it will hang at around 1.7 V instead of the expected 5V because the onboard LED and series resistor
 * pull the voltage level down, meaning it always returns LOW. If you must use pin PIN_LED as a digital input, use an
 * external pull down resistor.
 *
 * \param dwPin the pin number
 * \param dwVal HIGH or LOW
 */
extern void digitalWrite( uint32_t dwPin, uint32_t dwVal ) ;

/**
 * \brief Reads the value from a specified digital pin, either HIGH or LOW.
 *
 * \param ulPin The number of the digital pin you want to read (int)
 *
 * \return HIGH or LOW
 */
extern int digitalRead( uint32_t ulPin ) ;

#ifdef __cplusplus
}
#endif

#endif /* _WIRING_DIGITAL_ */

compulsivetinkerer, 2022-03-27:
I am still having this problem.

jmailloux, 2024-04-24:
@compulsivetinkerer @benmcmorran - I had this same problem and I figured it out.  It is because compilerArgs in c_cpp_properties.json has "--param" in it.  Removing that line in the compilerArgs list fixes the issue.  It is not a valid compiler argument.  The argument that is actually used is: --param max-inline-insns-single=500 .  The bug is the Arduino plugin doesn't generate the c_cpp_properties.json properly.

compulsivetinkerer, 2024-04-24:
Thanks.

I left the project for a a couple months and when I got back to it, it
started working.  I had assumed there was a software update.