microsoft/vscode-arduino #1453: usb-detection module fails to load

benmcmorran, 2022-02-09:
Hi, my VS Code 1.64.1 with Arduino extension v0.4.10 experienced the same behaviour: When clicked to open the serial monitor and reseted teh device (ESP8266 in my case), no output i shown. I have increased the Developer Log Level to 'trace' and ther is en exception related to this extension in `Log (Window)`Output - see attached log file [renderer1.log](https://github.com/microsoft/vscode-arduino/files/8036129/renderer1.log).

_Originally posted by @milanbx in https://github.com/microsoft/vscode-arduino/issues/1338#issuecomment-1034182766_

benmcmorran, 2022-02-10:
@milanbx What value to you have for the `arduino.enableUSBDetection` setting in VS Code? I think it's been broken for a while and should default to being disabled, but if you have it enabled you might see this issue.

[image]

milanbx, 2022-02-11:
UPDATE: I have turned the USB deteciotn off, upgraded to Code 1.64.2 and Arduino extension 0.14.1 and the Serial Monitor works fine!! Thanks for help!

adams13x13, 2022-02-11:
Dear Microsoft, are you serious? With almost every update of Visual Studio Code, the serial port stops working due to the incompatible Arduino extension and the update to the extension is not yet available. I had to disable VS Code updates on my machine because of this. Within one company (even that big as Microsoft), it should be possible to establish an arrangement between the teams ("VS Code" and "Arduino") to deliver the working product.

Indeed,
VS Code 1.64.2 + Arduino 0.4.10 does not work,
VS Code 1.64.2 + Arduino 0.4.11 works.

Tested with Linux Mint 19.3 (based on Ubuntu 18.04) Kernel: 5.4.0-99 (64bit) and ESP8266 / CH341 USB-to-Serial converter.

adams13x13, 2022-02-11:
@milanbx, what serial transmission rate is set in your sketch and in the terminal (VS Code)? Strange single characters could be indication of the different speed setting.
You can also try the "Arduino" serial monitor from Arduino IDE - does the hardware work correctly?

milanbx, 2022-02-11:
Good question, just to be sure, but the speed for serial is set to 115200: `Serial.begin(115200);` and the speed of terminal is set to the same. AndI have already tried Arduino IDE's serial monitor and ti works fine with the same sketch and board.

milanbx, 2022-02-11:
Just to give complete info, here is the simple sketch:

adams13x13, 2022-02-11:
@milanbx 

The line at 14:14:43.263 should be boot-loader output at 74880 baud. It is strange, but normal. ;-)

I have an impression, the serial monitor in VS Code starts to work properly about 2 seconds after "opening". Try to print something later (i.e. increase the delay of 500 ms to 3000) or print in the main loop like every second (not continuously). Also waiting between Serial.begin() and first Serial.println() could be useful. Or first activate the serial port in VS Code and then reset the ESP.

milanbx, 2022-02-11:
Well, I may try this, but every single sketch I have made until today (dozens) runs and prints to serial without any delays, even without the initial 500Ms like here!

milanbx, 2022-02-11:
OK, I have increased the initial delay to 1s and right after the sketch bin is uploaded and "Hard resetting via RTS pin..." done, the sketch starts and works fine (prints all lines as expected). However, the output is somehow snatchy, the lines are not printed at once. This is not appening in Arduino serial monitor, wher the output is fluent.
Moreover, everything described here is happening only to the first hardware reset. After that no output is given except "r$". And again, this does not happen in Arduino, where every reset just resets the sketch and prints the initial sequence over and over again. Strange...

benmcmorran, 2022-02-15:
@milanbx which version of the Arduino extension are you using? We made some major changes to the way serial works in version 0.4.11 that should help with handling reset, but please let me know if you're still seeing issues.

The "snatchy" behavior you see is likely related to #1412 which was required to avoid blocking the output window when there was a lot of data on the serial port. Does data appear roughly 3 times a second? That's the flushing interval for the output window.

milanbx, 2022-02-15:
I am using 0.4.11 Preview and it seems, that no more issues appear. You may consider this case closed for now. Thanks a lot for help.

benmcmorran, 2022-02-15:
Thanks for following up! Closing this issue.

closed 2022-02-15 by benmcmorran