microsoft/vscode-arduino #1458: Latest version breaks serial monitor - interferes with manual reset of ESP32.

clarkbremer, 2022-02-13:
With the latest version of the extension (0.4.11), the serial monitor interferes with a manual reset of my Arduino board (an ESP32).   It behaves as if the boot button is being held low (GPIO0 in my case).  So a manual reset causes it to enter boot mode instead of just running the code.

It works fine if I disconnect the USB cable.  And it works fine if I roll back to 0.4.10.

clarkbremer, 2022-02-17:
I'm using a pretty standard ESP32:  https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

The DTR line is wired to the "BOOT" input (GPIO0).  So if you reset the board, and this is low, it will enter "programming" mode, and give a message like this:  `rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download`.  The new version of your code holds this line low all the time.  So it will reboot after you program it, as it should, but any reset after that with the USB cable connected, and it will get stuck in "program" mode.

The only time DTR should go low is when you are trying to upload code to the device.

Also, Version 4.10 was also a regression from 4.9.  It does not make DTR go low at all.  So you have to do the equivalent with the button on the board.  I'm currently using 4.9, and it works properly.

benmcmorran, 2022-02-17:
Thanks for all that info! That should be enough for me to investigate this.

vlastahajek, 2022-02-21:
Additional info: Serial console should not control signals for reset and boot. Reboot is done by unloading. Opening the serial console shouldn't reset the device. Or, this could be at least configurable.

benmcmorran, 2022-03-23:
@clarkbremer @vlastahajek @hoekma Can you uninstall the Arduino extension and try installing the appropriate VSIX from [this CI build](https://github.com/microsoft/vscode-arduino/actions/runs/2031073763)? It includes [this change](https://github.com/microsoft/vscode-arduino/pull/1479) which should keep DTR, RTS, and CTS high while the serial monitor is active. It worked with my ESP32 Dev Module, but let me know if you see any issues.

clarkbremer, 2022-03-24:
Tested with ESP32 Dev Module and also an Arduino Pro Mini.  Both worked fine.

benmcmorran, 2022-03-24:
Thanks for verifying! I'll get this merged and in the next release.

vlastahajek, 2022-04-12:
I can confirm the fix works ok also for ESP8266 based boards.

gcampbell-msft, 2022-05-03:
This issue has been fixed in the latest release of this extension, which is available in the VS Code extension marketplace.

closed 2022-05-03 by gcampbell-msft

jamargevicius, 2023-05-18:
I'm using a ESP32-S3-WROOM device and find both the nRTS and nDTR lines are pulled low when I invoke the Serial Monitor after a code download.  I can download code fine when selecting the board "ESP32S3 Dev Module", but now have to select the board "ESP32 Dev Module" if I want to use the Serial Monitor.  This has to be a software bug somewhere. I can only find the ESP32-S3 support via the Arduino 1.8, but not the Arduino 2.  In Arduino 1.8, in File/Preferences/Additional Board Manager, I put: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json, https://dl.espressif.com/dl/package_esp32_index.json .... the github package has the S3 support.  In Arduino 2, I haven't located anything that supports the S3.