microsoft/vscode-arduino #1462: html-webpack Build Step Fails Node 17+

davidcooper1, 2022-02-17:
Right now the README specifies Node 14+. However, there is an error with the currently used version of `html-webpack-plugin` when used with Node 17. It results in a `digital envelope routines::unsupported` error. From what I've read, updating `webpack` and `html-webpack-plugin` versions may fix this issue.

Reference webpack/webpack#14532

gcampbell-msft, 2022-02-22:
Could be loosely related to #1442

gcampbell-msft, 2022-02-22:
@davidcooper1 If it's possible, a current workaround could be to downgrade your Node installation. However, we will try and look into this some in the near future and see if we can establish a possible longer-term solution.

fearthecowboy, 2022-02-28:
It would appear that webpack 4 breaks on Node 17+ -- the longer term solution would be to potentially upgrade to webpack 5 - but this is probably a bit of work.

Since Node 17 isn't a stable branch, and ... really there's not a lot of benefit to using node 17 during development anyway, I don't see that this is anything that needs to be instantly addressed.