microsoft/vscode-arduino #1465: [bug] Ardunio Board Config page shows （undefined），Cause analysis

VirensCn, 2022-02-26:
在boardManager.ts中，`loadInstalledPlatforms`函数中所调用的`getInstalledPlatforms`返回的IPlatform[]都缺少name属性

machine translation --->  In [boardManager.ts](https://github.com/microsoft/vscode-arduino/blob/a0a5f953dba56fc6d24447b1cd7fe588a1ee3072/src/arduino/boardManager.ts#L382), the iplatform [] returned by 'getinstalledplatforms' called in the' loadinstalledplatforms' function is missing the name attribute

[image]

最终导致在选择板子的时候出现undefined显示

machine translation --->  Finally, the undefined display appears when selecting the board
[image]

[BoardSelector.tsx](https://github.com/microsoft/vscode-arduino/blob/a0a5f953dba56fc6d24447b1cd7fe588a1ee3072/src/views/app/components/BoardSelector.tsx#L38)：
[image]

[/api/installedboards](https://github.com/microsoft/vscode-arduino/blob/a0a5f953dba56fc6d24447b1cd7fe588a1ee3072/src/arduino/arduinoContentProvider.ts#L40)
[image]

[getInstalledBoards](https://github.com/microsoft/vscode-arduino/blob/a0a5f953dba56fc6d24447b1cd7fe588a1ee3072/src/arduino/arduinoContentProvider.ts#L225)
[image]

aleun, 2022-03-18:
Thanks for your report! I wasn't able to reproduce this issue with the current version of the extension (v0.4.11). Here's what I see:

[image]

Could you please provide this information to help narrow down the issue?
- Extension version
- Arduino IDE version
- esp32 board package version

VirensCn, 2022-03-22:
I use  vscode-arduino version is 0.4.11 and use arduino-cli version is 0.21.1

[image]

my config 
[image]

VirensCn, 2022-03-27:
At present, I only use esp32, so I only know that this board has this problem