microsoft/vscode-arduino #1466: Can't open Arduino path

agnessap, 2022-03-06:
Hi, I'm trying to program an Arduino in VScode and I´ve specified the Arduino path in the JSON settings and I have installed the Arduino IDE, but VScode can still not find the path. 
I´ve checked multiple times, and the path seems to be correct.
Does anyone know how I can fix this issue? :)
[image]

BelKed, 2022-03-07:
It seems like you've set path of [`arduino-cli`](https://github.com/arduino/arduino-cli), not [Arduino IDE](https://github.com/arduino/Arduino).

In my case, binary of [`arduino-cli`](https://github.com/arduino/arduino-cli) is called `arduino-cli`, not just `arduino`, so that could be the problem. In that case, you should change `arduino.commandPath`.

But if you want to use [Arduino IDE](https://github.com/arduino/Arduino), you should change your `arduino.path`, because its version is probably not `0.21.1`...

You can find more info in [README.md](https://github.com/microsoft/vscode-arduino#options).

DeadlySurgeon, 2022-04-17:
> It seems like you've set path of [`arduino-cli`](https://github.com/arduino/arduino-cli), not [Arduino IDE](https://github.com/arduino/Arduino).
>
> In my case, binary of [...]

The documentation, as well as the errors I'm getting, tell me that the CLI needs to be specified under `arduino.path`.

[image]

But the funny thing is, this doesn't even work for me.

benmcmorran, 2022-04-29:
Hi @DeadlySurgeon. It looks like there is special case handling for finding the Arduino executable on Mac.

https://github.com/microsoft/vscode-arduino/blob/22eec0921e67d9922e505248293047ce83509d35/src/arduino/arduinoSettings.ts#L138-L143

https://github.com/microsoft/vscode-arduino/blob/22eec0921e67d9922e505248293047ce83509d35/src/common/util.ts#L442-L448

Based on this logic, you should be able to specify to specify the full path to `Arduino*.app` as `arduino.path`. Does that work for you?

DeadlySurgeon, 2022-04-30:
@benmcmorran 
>  It looks like there is special case handling for finding the Arduino executable on Mac.

This is on Linux.

closed 2022-06-06 by github-actions[bot]