microsoft/vscode-arduino #1467: Serial monitor breaks 1-wire protocol

Llaves, 2022-03-08:
The following code reads two DS18B20 digital temperature sensors and reports their values to the serial monitor.
```
// Program to read 1-wire temperatures and upload to ThingSpeak

#include "OneWire.h"
#include "DallasTemperature.h"

#define ONE_WIRE_BUS D3

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature.
DallasTemperature sensors(&oneWire);

void setup()
{
  Serial.begin(115200);
  sensors.begin();
  delay(1000);  //time for serial to connect
  Serial.print(sensors.getDeviceCount());
  Serial.println(" sensors found");
   Serial.print(sensors.getDS18Count());
  Serial.println(" DS18 sensors found");
 // first temp reading seems bogus, so skip one
  sensors.requestTemperatures(); // Send the command to get temperatures
  float sensor0 = sensors.getTempFByIndex(0);
  while ( sensor0 > 150)
  {
#ifdef DEBUG
    Serial.print("Discarding value ");
    Serial.println(sensor0);
#endif
    sensors.requestTemperatures(); // Send the command to get temperatures
    sensor0 = sensors.getTempFByIndex(0);
    delay(2000);
  }
}

void loop()
{
#ifdef DEBUG
  Serial.println("Requesting temperatures...");
#endif
  sensors.requestTemperatures(); // Send the command to get temperatures

  Serial.print("Temperature for Device 1 is: ");
  Serial.println(sensors.getTempFByIndex(0));

  Serial.print("Temperature for Device 2 is: ");
  Serial.println(sensors.getTempFByIndex(1));

  delay(5000);
}
```

I am testing this code on a Wemos D1 mini. When this code is compiled and uploaded from the Arduino IDE and observed via the Arduino IDE serial monitor, I get the expected results, showing the observed temperatures. If I compile and upload in VSCode and observe the output via the VSCode serial monitor, I get a result that indicates the devices aren't being read. If I compile/upload from VSCode but use the Arduino IDE serial monitor, I get the correct results, leading me to conclude that it is the serial monitor that is causing the problem. I have a more complex program that displays the temperature on an OLED display. The display shows the correct temperatures when there is no serial monitor or when connected to the Arduino IDE serial monitor. It displays the "no reading" value when connected to the VSCode serial monitor, providing additional evidence that the serial monitor is the root of the problem. I've tested with port speeds of 115200 and 19200 and get the same results.

Here's what "About" reports from VSCode
Version: 1.65.0 (user setup)
Commit: b5205cc8eb4fbaa726835538cd82372cc0222d43
Date: 2022-03-02T11:12:08.962Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19044

Arduino extension V0.4.11

Llaves, 2022-03-08:
I just stumbled on another case where the serial monitor is breaking the code, this time using the ESP_EEPROM library. The ESP_EEPROM_Simple.ino sample program produces no output on the VSCode serial monitor, works fine in the Arduino IDE serial monitor. 

```
// Example use of ESP_EEPROM library for ESP8266
//
// Normally writing to the 'emulated' EEPROM on ESP8266 requires an erase of the flash page used to hold
// the EEPROM data followed by a re-write of the changed data.
// The erasure takes a significant amount of time (10s of ms) and during this time
// interrupts must be blocked.
// In some cases this interferes with the sketch operation (e.g. it produces a noticeable
// blackout/flash for any PWM controlled lights as ESP8266 PWM relies on interrupts)
//
// The ESP_EEPROM library writes each new version of the EEPROM data to a new area until flash
// is full and so avoids wiping flash until necessary
//
// It's best for use when there are only a few things to save in EEPROM
// (i.e. total the size of the saved info is much smaller than the available flash size)
//

#include <ESP_EEPROM.h>

int eepromVar1 = 0;
long eepromVar2 = 1234;

void setup() {
  // Remember to set your serial monitor to 74880 baud
  // This odd speed will show ESP8266 boot diagnostics too
  Serial.begin(74880);
  Serial.println();

  // All the library functions are accessed via the EEPROM object created when
  // you include the library header ESP_EEPROM.h

  // The library needs to know what size you need for your EEPROM variables
  // The minimum size is 16

  // The begin() call is required to initialise the EEPROM library
  EEPROM.begin(16);

  // put some data into eeprom
  EEPROM.put(0, eepromVar1);  // int - so 4 bytes (next address is '4')
  EEPROM.put(4, eepromVar2);  // long - so 8 bytes (next address would be '12')

  // write the data to EEPROM
  boolean ok1 = EEPROM.commit();
  Serial.println((ok1) ? "First commit OK" : "Commit failed");

  // The eeprom data gets changed
  eepromVar1 = 1;   // Change some data

  // set the EEPROM data ready for writing
  EEPROM.put(0, eepromVar1);

  // commit (write) the data to EEPROM - only actually writes if there has been a change
  boolean ok2 = EEPROM.commit();
  Serial.println((ok2) ? "Second commit OK" : "Commit failed");

  // How to read stuff back into variables
  // variables should be same size as originally written
  int aNewVar1;
  long aNewVar2;
  EEPROM.get(0, aNewVar1);
  EEPROM.get(4, aNewVar2);
  Serial.print("Read back a variable 1 from EEPROM: ");
  Serial.println(aNewVar1);
  Serial.print("Read back a variable 2 from EEPROM: ");
  Serial.println(aNewVar2);

}

void loop() {
  delay(1000);
}
```

same setup as above

benmcmorran, 2022-03-23:
@Llaves Can you uninstall the Arduino extension and try installing the appropriate VSIX from [this CI build](https://github.com/microsoft/vscode-arduino/actions/runs/2031073763)? It includes [this change](https://github.com/microsoft/vscode-arduino/pull/1479) which should keep DTR, RTS, and CTS high while the serial monitor is active. I'm not sure if that will fix your issue, but it seems similar to #1458.

Llaves, 2022-03-27:
I'm out of town. I'll be able to test this in a few days when I have all the necessary parts in front of me.

Llaves, 2022-03-29:
That seems to have fixed the problem, at least for the two code examples listed here.

benmcmorran, 2022-03-29:
Great! Thanks for letting me know.

closed 2022-05-03 by github-actions[bot]