microsoft/vscode-arduino #1468: Basic `wiring` identifiers are undefined?

zfields, 2022-03-08:
[image]

As you can see in the screenshot, VSCode claims the most basic wiring APIs are not defined.

I have validated the paths in the `c_cpp_preferences.json`, and they are correct. I also noticed that the problem matcher claims it is checking against C/C++(20) in the **PROBLEMS** tab. However, the standard specified in `c_cpp_preferences.json` clearly states `c++14`.
_For the record: The `c_cpp_preferences.json` file was generated by the extension. I did NOT modify the file; I only verified its contents._

The code compiles without warnings, and if I press F12, then it will find the definition of either of the API.

benmcmorran, 2022-03-24:
Hi @zfields! I'm was able to reproduce this issue, although eventually the identifiers did resolve. Given that F12 works, I suspect this is an issue with the C++ extension that powers our IntelliSense. I'm going to transfer this issue to the cpptools repo for further investigation.

benmcmorran, 2022-03-24:
Sorry, looks like I don't have permission to transfer. Can you [open an issue with this template](https://github.com/microsoft/vscode-cpptools/issues/new?assignees=&labels=&template=language-service.md&title=)?

closed 2022-03-24 by benmcmorran