microsoft/vscode-arduino #1469: Timestamp format does not work for %f

iqapps, 2022-03-08:
I tried entering a format "%f" in the timestamp format field, after clicking the watch on the Statusbar, namely the microseconds, but it does not work for me on Windows.

Entering %f just prints f
Entering %F prints the date - but this is not even documented, as I can see it, but it means that it does work to enter something.
Entering %X prints the time with seconds as expected, but I need more detailed times than that.

BelKed, 2022-03-08:
This is because this project recently moved to a different serial port library that uses a different library for the time handling than the previous one.
You can find an updated list of all available placeholders [here](https://github.com/samsonjs/strftime#supported-specifiers).

---

I will also submit a pull request in a few minutes, so this change will be reflected in the `README.md` file and, if necessary, elsewhere.

BelKed, 2022-03-09:
You're welcome :)

aleun, 2022-03-18:
The docs were updated to reflect the new format in #1470 (thanks, @BelKed!) and will ship in the next release.

gcampbell-msft, 2022-05-03:
This issue has been fixed in the latest release of this extension, which is available in the VS Code extension marketplace.

closed 2022-05-03 by gcampbell-msft