microsoft/vscode-arduino #1471: command 'arduino.showBoardConfig' not found

carreylife, 2022-03-11:
macOS Mojave 
version 10.14.6

vscode 1.65
 settings.json
[image]

aleun, 2022-03-18:
Thanks for your report. What version of the extension, and what version of the Arduino IDE do you have installed?

Also, can you try setting `arduino.commandPath` to `Contents/MacOS/Arduino`? `arduino_debug.exe` would be the default setting for Windows, not Mac.

carreylife, 2022-03-21:
> Thanks for your report. What version of the extension, and what version of the Arduino IDE do you have installed?
> 
> Also, can you try setting `arduino.commandPath` to `Contents/MacOS/Arduino`? `arduino_debug.exe` would be the default setting for Windows, not Mac.

Thanks for your reply.
**vscode extension: 0.4.6 preview**
**Arduino IDE: 1.8.19**
i have setting arduino.commandPath to Contents/MacOS/Arduino but not work.

benmcmorran, 2022-03-24:
Hi @carreylife. The current release of the extension is version 0.4.11, not 0.4.6. Can you try updating and let me know if you still see the issue?

Also to clarify, can you describe the steps you are taking, what you expect see, and what actually happens?

MonveChen, 2022-04-15:
Me too, it took a long time and finally gave up

nya3jp, 2022-04-15:
Hi,

I encountered the same error message for every Arduino extension commands. After a bit of debugging, I found that Arduino extension fails to activate if the setting `C_Cpp.intelliSenseEngine` is disabled. The extension started working once I enabled the setting.

Hope this helps.

benmcmorran, 2022-04-29:
Thanks for that clue @nya3jp. Unfortunately, I'm still not able to reproduce this issue. I tried adding this line to a workspace settings.json file.

```
    "C_Cpp.intelliSenseEngine": "Disabled"
```

However, the Arduino extension still appeared to activate correctly and board management commands worked. Am I missing any steps?

nya3jp, 2022-04-30:
Looks like the issue does not happen with latest C/C++ extension (v1.9.8). I can reproduce the issue on v1.9.7, but not on v1.9.8.

closed 2022-06-06 by github-actions[bot]

Nebula83, 2023-01-15:
Settings that worked for me:
command path: `Contents/MacOS/Arduino`
Arduino path: `/Applications/Arduino.app`

but YMMV