microsoft/vscode-arduino #1473: Why can't `arduino.json` "output" be the same as the output folder generated for debugging?

zfields, 2022-03-11:
I LOVE the debugging feature of VSCode for Arduino!

When I debug, instead of using a `/tmp` folder, it automatically creates the folder `${workspaceRoot}/.build/BluesW` - I love it.

What I don't love, is that I can't use that same folder for the `output` key in my `arduino.json`. If I specify `.build/BluesW` to match the folder that has already been created, then it will put the binaries in `.build/BluesW/BluesW`. However, when I simply specify `.build` it will place the binaries in the root of the `.build` folder (as you would expect). :shrug: 

Also, if I completely remove the `.build` folder, but I have a nested path (e.g. `.build/BluesW`) specified for `output`, then it throws an error that the directory does not exist.

Here is my `arduino.json` file:

`arduino.json`
{
    "configuration": "pnum=SWAN_R5,upload_method=swdMethod,xserial=generic,usb=CDCgen,xusb=FS,opt=ogstd,dbg=enable,rtlib=nano",
    "board": "STMicroelectronics:stm32:BluesW",
    "port": "/dev/ttyACM0",
    "sketch": "sfy-bouy.ino",
    "output":".build/BluesW/"
}

gcampbell-msft, 2022-03-31:
@zfields Could you share your test project including your debug configuration (launch.json if you have one) so that I can attempt to repro? Thanks!

zfields, 2022-03-31:
What board do you have? I'll try and tailor a project just for you.

IIRC, you can just use the extension to create the blink sketch, and try and debug it and it will cause the issue to repro when it generates the build. No board required.

gcampbell-msft, 2022-03-31:
I'm using one of the STM Discovery Boards, STMicroelectronics B-L4S5I-IOTOA1. 

My main question is what your debugging configuration looks like.

zfields, 2022-03-31:
For the most part, it's the one that gets generated by the extension. I'm using OpenOCD as well.

Here are the full instructions.
https://dev.blues.io/quickstart/swan-quickstart/#debugging-arduino-with-vs-code

gcampbell-msft, 2022-03-31:
Great. I'll attempt to repro as soon as my bandwidth allows and try to make sure this is reproducible. 

I should also mention that the experience you're describing is very likely by design (I'll look into this along with attempting to reproduce), as build files often get overwritten and its often better practice to keep them separate.

gcampbell-msft, 2022-04-01:
@zfields Understood. What version of the extension are you running? I can confirm that I'm reproducing all of the scenarios you laid out except for the first where it creates a /tmp folder and I don't see anywhere in the code that creates a folder like that. 

I've published a PR that attempts to fix most of these issues (the ones listed in 1 and 2). If you download the VSIX from this PR and test with it, could you confirm that it solves your issues? [Here ](https://github.com/microsoft/vscode-arduino/actions/runs/2078521671)is the link where the VSIX's are located.

zfields, 2022-04-01:
Ahhh... I'm on a Linux machine. `/tmp` is at the root of my file system and is just a temporary directory that gets deleted when I reset my computer. It's not something I can see in VSCode or anything. I only know it's getting created from the compiler spew in the logs.

To recount, if I don't specify an `"output":` folder, then I get this warning...

```
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
```

and I will eventually see this from the compiler output...

```
...
Linking everything together...
/home/zak/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/bin/arm-none-eabi-gcc -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Og -g --specs=nano.specs -Wl,--defsym=LD_FLASH_OFFSET=0 -Wl,--defsym=LD_MAX_SIZE=2097152 -Wl,--defsym=LD_MAX_DATA_SIZE=655360 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--default-script=/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/variants/STM32L4xx/L4R5Z(G-I)Y_L4R9Z(G-I)Y_L4S5ZIY_L4S9ZIY/ldscript.ld -Wl,--script=/home/zak/.arduino15/packages/STMicroelectronics/hardware/stm32/2.2.0/system/ldscript.ld -Wl,-Map,/tmp/arduino_build_932108/sfy-bouy.ino.map -L/home/zak/.arduino15/packages/STMicroelectronics/tools/CMSIS/5.7.0/CMSIS/DSP/Lib/GCC/ -larm_cortexM4lf_math -o /tmp/arduino_build_932108/sfy-bouy.ino.elf -L/tmp/arduino_build_932108 -Wl,--start-group /tmp/arduino_build_932108/sketch/SrcWrapper.cpp.o /tmp/arduino_build_932108/sketch/sfy-bouy.ino.cpp.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/NoteI2c_Arduino.cpp.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/NoteLog_Arduino.cpp.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/NoteSerial_Arduino.cpp.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/Notecard.cpp.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_atof.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_b64.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_cjson.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_cjson_helpers.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_const.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_ftoa.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_helpers.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_hooks.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_i2c.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_md5.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_printf.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_request.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_serial.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_str.c.o /tmp/arduino_build_932108/libraries/Blues_Wireless_Notecard/note-c/n_ua.c.o /tmp/arduino_build_932108/libraries/Wire/Wire.cpp.o /tmp/arduino_build_932108/libraries/Wire/utility/twi.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/syscalls.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_adc.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_adc_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_can.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_cec.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_comp.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_comp_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_cordic.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_cortex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_crc.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_crc_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_cryp.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_cryp_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dac.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dac_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dcache.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dcmi.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dcmi_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dfsdm.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dfsdm_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dma.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dma2d.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dma_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dsi.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_dts.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_eth.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_eth_ex.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_exti.c.o /tmp/arduino_build_932108/libraries/SrcWrapper/HAL/stm32yyxx_hal_fdcan.c.o /tmp/arduino_build_932108/libr [... cut ...]
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
[Done] Verifying sketch 'sfy-bouy.ino'
```

Check the end of the 5th line from the bottom:
```
/home/zak/.arduino15/packages/STMicroelectronics/tools/xpack-arm-none-eabi-gcc/10.3.1-2.3/bin/arm-none-eabi-size -A /tmp/arduino_build_932108/sfy-bouy.ino.elf
```

You can see the output goes into `/tmp/arudino_build_932108/`

Thanks for all the help! I will test your .vsix and get back to you.

zfields, 2022-04-01:
**I love it, it works perfectly!**

The only other thing I would to see change is to add ...

```json
"output": ".build"
```

... to the `arduino.json` by default.

You have already chosen it as the path for the debug binaries, and this would shine light on where someone might expect to find them. Also, I don't understand the value of not caching the build output (e.g. using `/tmp/arduino_build_...` instead), since you are already placing binaries on someone's file system; albeit randomly. Plus if you needed to clear them away completely (for sanity), then you can just delete the `.build` folder it is very prominently featured in VSCode file system view (very near the top).

benmcmorran, 2022-04-29:
Reopening because the fix is merged but not yet released.

closed 2022-05-03 by github-actions[bot]