microsoft/vscode-arduino #1476: Upload and Verify buttons are missing from the Actionbar

iqapps, 2022-03-16:
Not sure when this was implemented, but can we have them back or configurable.

benmcmorran, 2022-03-23:
Hi @iqapps! Can you clarify what you mean by the "Actionbar"? Using the [terminology from this article](https://code.visualstudio.com/docs/getstarted/userinterface), there should already be upload and verify buttons in the "editor toolbar", but there are no upload and verify buttons in the "status bar". 

[image]

iqapps, 2022-03-24:
Hi, It is a typo, sorry. I ment to write "editor toolbar" :-)

I can see you have the Arduino Upload and verify buttons, but I dont. I have GIT buttons instead.

[image]

When I search in settings with various keywords, anything found, nothing relates to make the Arduino extension show these buttons.

iqapps, 2022-03-24:
Ahhhh, really ! 

Its only showing the buttons when an actual INO file is selected.

gcampbell-msft, 2022-03-31:
@iqapps Based on your most recent comment, it seems you found what is going on based on the behavior of those buttons only showing when an INO file is selected. Closing.

closed 2022-03-31 by gcampbell-msft