microsoft/vscode-arduino #1477: Support Arduino IDE 2.0

letmaik, 2022-03-23:
https://www.arduino.cc/en/software#experimental-software

benmcmorran, 2022-03-24:
Thanks for the suggestion. We don't have capacity to immediately add this support, but I'm marking this issue as feature request for tracking and so that others can vote on it.

azinman, 2022-07-12:
My experience is that the Arduino ESP32 integration on M1 Macs doesn't work well on 1.8, but it does on 2.0. Please consider adding sooner, or at least support with the cli tool. Switching between vscode & Arduino is a bit painful, and Arduino's text editor is elementary.

integrated-circuit, 2022-09-09:
Agreed @azinman, it's better than the 1.X version

fra87, 2022-10-01:
Arduino 2.0 came out of beta and is going to spread more and more; I think it is becoming more and more important to support it, since a lot of people will start using it

arminrosu, 2022-10-14:
> Please consider adding sooner, or at least support with the cli tool.

The extension works fine with Arduino IDE 2.0 installed, _but_ using `arduino-cli`. That's how I got it working.

Just `brew install arduino-cli` then follow [this guide](https://medium.com/@thomas.kilmar/arduino-cli-with-visual-studio-code-on-macos-d2ad32ff0276) to setup the extension.

> I think it is becoming more and more important to support it, since a lot of people will start using it

Agreed. _But_, it looks like v2.0 of the IDE does not contain any header files to reference (on Mac). v1 did. `arduino-cli` might be the only way to use v2.0 with the extension.

ChrisKader, 2022-10-24:
Would be good to update the README to indicate that v2.0 of the IDE is not supported.

### [Arduino IDE](https://github.com/Microsoft/vscode-arduino#arduino-ide)
The Arduino IDE can be installed the Arduino [download page](https://www.arduino.cc/en/main/software#download).

- The supported Arduino IDE versions are 1.6.x and later.
- The Windows Store's version of the Arduino IDE is not supported because of the sandbox environment that the application runs in.
- Note: Arduino IDE 1.8.7 had some breaking changes, causing board package and library installation failures. These failures were corrected in 1.8.8 and later.

LazaroFilm, 2023-01-04:
Aruino IDE 2.0 not being supported is a real annoyance. It's now out of beta (currently 2.0.3) and still not supported...

laurensV, 2023-01-13:
Agreed, would be great if this would also work with Arduino IDE 2.X

atypic, 2023-01-13:
The instructions are now broken.

manolisss, 2023-01-25:
Hello, any timeline on the integration of Arduino IDE 2.x in VS Code? Thanks

arminrosu, 2023-01-25:
@benmcmorran as far as I can tell, Arduino IDE 2 cannot be supported by the extension, due to lack of the command line tools. Would it make sense to drop IDE support alltogether, to avoid confusion?

LazaroFilm, 2023-01-25:
I see in that case promoting the ClI option as the primary install method may be the best route.

benmcmorran, 2023-02-16:
Thanks for the feedback. [Only the legacy IDE and Arduino CLI are currently officially supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

gzwz, 2023-03-02:
Please consider adding sooner

closed 2023-04-08 by github-actions[bot]

hlide, 2024-08-24:
Oh great... the Arduino 2.x.x editor, even if it's better than the 1.8.x one, is still a big mess to use. I discovered that there was an extension and now I learn that it doesn't support 2.x.x and that the extension is obsolete. Count me in the ‘Arduino IDE sucks’ club.

LazaroFilm, 2024-09-03:
I have moved to PlatformIO and never looked back.