microsoft/vscode-arduino #1478: Activation Event `*` causes `ms-vscode.cpptools` to load in every directory

daniel-stoneuk, 2022-03-23:
I was encountering extremely high (100%) CPU usage from the [ms-vscode.cpptools](https://github.com/microsoft/vscode-cpptools) `cpptools` process and determined that it was due to that extension indexing every single file within my home directory. This would go on continuously, and even after VSCode was closed. I'll be making an issue over there for why the process continued after the VSCode process ended.

Since the Arduino extension is set to activate whenever VSCode starts up, and is dependent on the C++ extension, an empty VSCode window with no open project would cause this issue to occur. For this reason I have had to disable the Arduino extension.

Would it be possible to reduce the scope of the [activation events](https://code.visualstudio.com/api/references/activation-events) to only run when relevant files are found or commands run?

daniel-stoneuk, 2022-03-24:
An update from the linked issue in vscode-cpptools: the high CPU usage was due to a misconfiguration on my machine. However, I believe this would still be a good change to make.

BelKed, 2022-03-24:
I agree with this change. It takes about half a second to load this extension with `vscode-cpptools`.
This may seem like a negligible time, but these two extensions are loading the slowest of the all installed extensions...

[image]

benmcmorran, 2022-03-24:
Fixed by #1481.

closed 2022-05-03 by github-actions[bot]