microsoft/vscode-arduino #1485: Some compile errors dont show, as compiler exit with code 1

iqapps, 2022-03-25:
Having the Arduino Extension Log Level set to "info", and running Verify has compilation errors, somestimes the compiler just outputs "Exit 1" with no trace of the code error. 

To show the code error, i need to switch to Verbose mode and run Verify again. Then after fixing one or more errors, I can switch back to Info mode, and then compile again, and for the most part, see detailed error descriptions for editing code errors.

There are a few error types that typically (but not always) causes the "Exit 1" on the output, while others show the error details in "Info" mode. Declaration errors, after editing usually causes the "Exit 1".

I have made a little example with my production code. The entire code compiles without errors, then I have edited, in one place, by replacing "String" with "string", and now Verify causes the "Exit 1" output,

```
    string reply = Format(F("LS00%03X%c%c%04X%X"), door, open, limit, current, errorCode);
```

[Starting] Verifying sketch 'LockController_3.ino'
Please see the build logs in output path: c:\Code\Biometric2\Hardware\Arduino\_builds\LockController_3
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
**exit status 1**
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Ctrl+Alt+I"
[Error] Verifying sketch 'LockController_3.ino': Exit with code=1

I can Verify or try Upload many times and every time I get the "Exit 1".

_By the unstable nature of the problem, I suspect this to be a timing issue, so, I have pasted my setup below:_

Device name	BIOCLI17
Full device name	BIOCLI17.bio.bio
Processor	Intel(R) Core(TM) i9-9900 CPU @ 3.10GHz   3.10 GHz
Installed RAM	32,0 GB (31,8 GB usable)
Device ID	80A3A898-1272-4991-A02D-7830127657A5
Product ID	00330-53059-80649-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

My workspace has this many lines:
```
total files : 13
   total code lines : 3855
   total comment lines : 955
   total blank lines : 633
   |      extension|     total code|  total comment|    total blank|percent|
   -------------------------------------------------------------------------
   |           .ino|           1618|            364|            239|     42|
   |             .h|           2231|            589|            393|     58|

```

gcampbell-msft, 2022-03-31:
What version of the extension are you using?
I attempted to use the same pattern you did to repro the issue. The pattern of replacing "String" with "string" and I received output like the following:
[image]
This was done with log level `info`.

iqapps, 2022-04-04:
Nice, but as I said, sometimes this does not happen.
It looks to me, as a race condition.