microsoft/vscode-arduino #1488: Intellisense not doing complete updating

iqapps, 2022-04-04:
I have intellisense report missing references in my workspace, for some keywords, while most other works fine.

Getting this error when pressing F12, While the called class clearly contains the method, and have had it for several iterations...

[image]

Even pressing F12 on the method itself, says na...dont know what this is... :-)

[image]

This is just one example, but I experience it throughout the workspace, but at somewhat rare occasions luckily. This problem has existed for several versions of the extension.

Rebuilding intellisense with CTRL+ALT+I says its doing something, but this do not solve the problem.
Restarting VSCode dont help.

I would love to give you some logs, but they are not where the verify output says it should be. Even when Verbose output is turned on.

`
Please see the build logs in output path: c:\Code\.....
`

nimish, 2022-04-17:
Same here. Mac OS, arduino-cli. Include paths generated are missing the `/**` recursion.

benmcmorran, 2022-04-29:
Thanks for the report @iqapps. IntelliSense in this extension works by generating a `c_cpp_properties.json` file in the `.vscode` directory. Can you share that file with us after removing any sensitive information? What we're looking for is that the path to the file you are editing is included in the configuration file somewhere, which should include it for IntelliSense processing.

closed 2022-06-05 by github-actions[bot]