microsoft/vscode-arduino #1489: Failed to generate IntelliSense configuration with STM8 Core

0x0fe, 2022-04-09:
To reproduce the issue, install STM8 core https://github.com/stm32duino/Arduino_Core_STM8 from the board manager, select any example, one of the two boards as target, and hit "Verify".

```
Verification...
[Warning] Failed to generate IntelliSense configuration.
```

Is there a way to get some log? if i got the error i can fix it, but here there is logging of the error.

BelKed, 2022-04-10:
You can get more verbose log by setting `arduino.logLevel` to `verbose`:

```json
{
  "arduino.logLevel": "verbose"
}
```

0x0fe, 2022-04-10:
well, except it doesnt work.

[image]

0x0fe, 2022-04-26:
> same +1

i guess we need to fix it ourselves, since microsoft nor ST seem interrested to help

nya3jp, 2022-04-28:
You may want to try https://github.com/microsoft/vscode-arduino/pull/1494. It fixes the issue for m5stack for me.

0x0fe, 2022-04-28:
yes, i did not think of it, but yes, it is a common problem indeed, 'll give a try and not thanks microsoft nor ST for the non existing support

0x0fe, 2022-04-28:
ST, a company that probably lost a huge percentage of market share with the disaster of last month, still did not understood the importance of support.i feel like its going to be "good ridance" soon

benmcmorran, 2022-04-29:
Hi @0x0fe. I'm uable to reproduce this issue because I don't have a license for the compiler used by the STM8 tools. However, based on the earlier discussion #1494 may fix your issue, which will be included in an upcoming release. You can try it now by installing the VSIX manually from [this CI build](https://github.com/microsoft/vscode-arduino/actions/runs/2247797810). Does that fix your issue?

[image]

0x0fe, 2022-04-30:
> Hi @0x0fe. I'm uable to reproduce this issue because I don't have a license for the compiler used by the STM8 tools. However, based on the earlier discussion #1494 may fix your issue, which will be [...]

Thank you for checking on the issue. I am going to try it soon when i have a chance to get back at this design, regarding the STM8 compiler "license", I think you can request one anytime here (it is automated via email) : 
C:\Users\user\AppData\Local\Arduino15\packages\STMicroelectronics\tools\cxppstm8\4.1.3\RegArdui.exe

0x0fe, 2022-04-30:
[image]
Since there is a VSIX i gave a try, installed the VSIX from the contextual menu of extension pane on left side. Now the the "inderterminate" progress bar is running for a good 10 minutes, it does not seem to do anything, no progress report of any kind anywhere neither, just stuck doing nothing. So i guess i will not be able to test wether the fix works.

As a side note, when the user initiates an action such as installing an extension he expects a feedback on what is going on, and wether it failed or succeeded, and that a timeout will prevent the whole application be be stuck forever if the action fails, pretty basic UI concept, but not obvious for everyone apparently.

0x0fe, 2022-04-30:
I had to quit completely VScode (force quit) then i restarted it, retried the VSIX installing, and now it worked. This kidn of buggy inconsistence is pretty consistent with microsoft style i would say.

0x0fe, 2022-04-30:
well, sadly it not only did not solve the problem, it also completely broke the extension, the "verify" command does not exist anymore, cannot even build.

It is also burning a lot of CPU cycles when i open the directory, without any report in OUTPUT, it is quite not acceptable to have sudden huge CPU usage without any justification or report.

[image]

0x0fe, 2022-04-30:
[image]

benmcmorran, 2022-04-30:
Thanks for letting me know. We found some breaking issues in the release
candidate (unrelated to this particular fix) that are causing activation to
fail, which we hope to fix next week.

benmcmorran, 2022-06-06:
Hi @0x0fe. v0.4.12 of the extension available on the VS Code marketplace, which fixes the activation issues described above. Let me know if you are still having issues with STM8 development. Otherwise, this issue will be automatically closed.

closed 2022-07-13 by github-actions[bot]