microsoft/vscode-arduino #1490: Arduino IDE Splash Screen when verifying/uploading

fussel132, 2022-04-09:
Hey guys,
I've been researching that now since a while and wasn't able to find a solution for my problem. On my Mac, when I verify or upload my sketch, I get a lot of debug output that comes in hand with the Arduino IDE Splash Screen showing up while compiling. My Log Level however is set to "arduino.logLevel": "info" but I still get this mass of verbose output when compiling:

> [Starting] Verifying sketch 'secretproject.ino'
Please see the build logs in output path: /Users/user/Library/CloudStorage/OneDrive-Persönlich/Dokumente/Arduino/secretproject/.build
2022-04-09 22:48:46.188 Arduino[55373:2073794] Loading Application 'Arduino'
2022-04-09 22:48:46.189 Arduino[55373:2073794] JVMRuntime=jre8u252-b09.jre
2022-04-09 22:48:46.190 Arduino[55373:2073794] CFBundleName=Arduino
2022-04-09 22:48:46.190 Arduino[55373:2073794] WorkingDirectory=(null)
2022-04-09 22:48:46.190 Arduino[55373:2073794] JVMMainClassName=processing.app.Base
2022-04-09 22:48:46.190 Arduino[55373:2073794] JVMOptions=(
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
2022-04-09 22:48:46.190 Arduino[55373:2073794] JVMArguments=(
)
2022-04-09 22:48:46.190 Arduino[55373:2073794] JVMClasspath=(null)
2022-04-09 22:48:46.190 Arduino[55373:2073794] JVMDefaultOptions={
}
2022-04-09 22:48:46.190 Arduino[55373:2073794] SearchSystemJVM=false
2022-04-09 22:48:46.190 Arduino[55373:2073794] -> Bundle path: /Applications/Arduino.app
2022-04-09 22:48:46.190 Arduino[55373:2073794] -> Working Directory: '/Users/user/Library/CloudStorage/OneDrive-Persönlich/Dokumente/Arduino/secretproject'
2022-04-09 22:48:46.190 Arduino[55373:2073794] -> JVM Runtime path: /Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre
2022-04-09 22:48:46.190 Arduino[55373:2073794] -> Java Runtime Dylib Path: '/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/lib/jli/libjli.dylib'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Command line passed to application argc=28:
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 0: '/Applications/Arduino.app/Contents/MacOS/Arduino'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 1: '-Djava.class.path=/Applications/Arduino.app/Contents/Java/rsyntaxtextarea-3.0.3-SNAPSHOT.jar:/Applications/Arduino.app/Contents/Java/batik-ext-1.8.jar:/Applications/Arduino.app/Contents/Java/slf4j-api-1.7.22.jar:/Applications/Arduino.app/Contents/Java/arduino-core.jar:/Applications/Arduino.app/Contents/Java/batik-xml-1.8.jar:/Applications/Arduino.app/Contents/Java/slf4j-simple-1.7.22.jar:/Applications/Arduino.app/Contents/Java/batik-dom-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-compress-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-httpclient-3.1.jar:/Applications/Arduino.app/Contents/Java/bcprov-jdk15on-152.jar:/Applications/Arduino.app/Contents/Java/commons-logging-1.0.4.jar:/Applications/Arduino.app/Contents/Java/batik-script-1.8.jar:/Applications/Arduino.app/Contents/Java/jackson-annotations-2.9.5.jar:/Applications/Arduino.app/Contents/Java/batik-parser-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-squiggle-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-awt-util-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-gvt-1.8.jar:/Applications/Arduino.app/Contents/Java/bcpg-jdk15on-152.jar:/Applications/Arduino.app/Contents/Java/xml-apis-1.3.04.jar:/Applications/Arduino.app/Contents/Java/batik-anim-1.8.jar:/Applications/Arduino.app/Contents/Java/jtouchbar-1.0.0.jar:/Applications/Arduino.app/Contents/Java/batik-bridge-1.8.jar:/Applications/Arduino.app/Contents/Java/jssc-2.8.0-arduino4.jar:/Applications/Arduino.app/Contents/Java/batik-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-transcoder-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-svg-dom-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-rasterizer-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-codec-1.8.jar:/Applications/Arduino.app/Contents/Java/commons-io-2.6.jar:/Applications/Arduino.app/Contents/Java/commons-codec-1.7.jar:/Applications/Arduino.app/Contents/Java/xmlgraphics-commons-2.0.jar:/Applications/Arduino.app/Contents/Java/apple.jar:/Applications/Arduino.app/Contents/Java/commons-exec-1.1.jar:/Applications/Arduino.app/Contents/Java/batik-css-1.8.jar:/Applications/Arduino.app/Contents/Java/batik-util-1.8.jar:/Applications/Arduino.app/Contents/Java/jna-platform-4.2.2.jar:/Applications/Arduino.app/Contents/Java/commons-net-3.3.jar:/Applications/Arduino.app/Contents/Java/batik-svgpp-1.8.jar:/Applications/Arduino.app/Contents/Java/xml-apis-ext-1.3.04.jar:/Applications/Arduino.app/Contents/Java/jmdns-3.5.5.jar:/Applications/Arduino.app/Contents/Java/java-semver-0.8.0.jar:/Applications/Arduino.app/Contents/Java/pde.jar:/Applications/Arduino.app/Contents/Java/jna-4.2.2.jar:/Applications/Arduino.app/Contents/Java/commons-lang3-3.8.1.jar:/Applications/Arduino.app/Contents/Java/jackson-core-2.9.5.jar:/Applications/Arduino.app/Contents/Java/jsch-0.1.50.jar:/Applications/Arduino.app/Contents/Java/jackson-databind-2.9.5.jar'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 2: '-Djava.library.path=/Applications/Arduino.app/Contents/MacOS'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 3: '-DLibraryDirectory=/Users/user/Library'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 4: '-DDocumentsDirectory=/Users/user/Documents'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 5: '-DApplicationSupportDirectory=/Users/user/Library/Application Support'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 6: '-DCachesDirectory=/Users/user/Library/Caches'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 7: '-DSandboxEnabled=true'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 8: '-Dapple.awt.application.name=Arduino'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 9: '-Dcom.apple.macos.use-file-dialog-packages=true'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 10: '-Dcom.apple.smallTabs=true'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 11: '-DAPP_DIR=/Applications/Arduino.app/Contents/Java'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 12: '-Djava.ext.dirs=/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/lib/ext/:/Applications/Arduino.app/Contents/PlugIns/jre8u252-b09.jre/Contents/Home/jre/lib/ext/'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 13: '-Djava.net.preferIPv4Stack=true'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 14: '-Xdock:name=Arduino'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 15: '-Dcom.apple.mrj.application.apple.menu.about.name=Arduino'
2022-04-09 22:48:46.201 Arduino[55373:2073794] Arg 16: '-Dfile.encoding=UTF-8'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 17: '-Xms128M'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 18: '-Xmx512M'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 19: '-splash:/Applications/Arduino.app/Contents/Java/lib/splash.png'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 20: 'processing.app.Base'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 21: '--board'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 22: 'arduino:avr:nano:cpu=atmega328old'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 23: '--verify'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 24: '--verbose-build'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 25: '--pref'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 26: 'build.path=/Users/user/Library/CloudStorage/OneDrive-Pers√∂nlich/Dokumente/Arduino/secretproject/.build'
2022-04-09 22:48:46.202 Arduino[55373:2073794] Arg 27: '/Users/user/Library/CloudStorage/OneDrive-Pers√∂nlich/Dokumente/Arduino/secretproject/secretproject.ino'
2022-04-09 22:48:46.202 Arduino[55373:2073803] Loading Application 'Arduino'
2022-04-09 22:48:46.203 Arduino[55373:2073803] JVMRuntime=jre8u252-b09.jre
2022-04-09 22:48:46.203 Arduino[55373:2073803] CFBundleName=Arduino
2022-04-09 22:48:46.203 Arduino[55373:2073803] WorkingDirectory=(null)
2022-04-09 22:48:46.203 Arduino[55373:2073803] JVMMainClassName=processing.app.Base
[... cut ...]
Wenig Arbeitsspeicher verfügbar, es können Stabilitätsprobleme auftreten.
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Cmd+Alt+I"
[Done] Verifying sketch 'secretproject.ino'

Would be cool if someone could tell me which setting I have to change in order to get rid of that output.
On my other pc I do not get that output or the Splash Screen when using VSCode to compile, so it should be a setting or parameter that unintentionally changed :/

Thanks for the help in advantage!

BelKed, 2022-04-10:
When verifying or compiling on MacOS, I get almost identical output and the splash screen is also displayed.
What is the OS on the second PC?

benmcmorran, 2022-04-29:
Thanks for the report @fussel132! I think the verbose output is happening because we need to update the list regexes for verbose output. I opened #1498 to fix the issue based on the log you provided. If you'd like to test it, you can uninstall the extension, then [manually install it from a VSIX in this build](https://github.com/microsoft/vscode-arduino/actions/runs/2247085744).

I don't think we have much control over whether the splash screen is shown. If it's really bothering you, you can try using the arduino-cli instead of the Arduino IDE, which should launch silently in the background.

https://github.com/microsoft/vscode-arduino/blob/3777ed761af1c46963367f720b7016021876c61c/src/arduino/arduino.ts#L781-L785

BelKed, 2022-04-29:
> I don't think we have much control over whether the splash screen is shown. If it's really bothering you, you can try using the arduino-cli instead of the Arduino IDE, which should launch silently in the background.

Well, if I see correctly, there's one line that points to the splash image (`-splash:$APP_ROOT/Contents/Java/lib/splash.png`)

```
…
2022-04-29 21:57:02.431 Arduino[60682:800028] JVMOptions=(
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
…
```

However, I don't know where is this command launched from. But maybe removing this `splash` would fix this issue...

benmcmorran, 2022-04-29:
Yeah, I don't see the string "splash" anywhere in the vscode-arduino codebase, so I'm guessing it's passed internally to the JVM as part of the Arduino application. It looks like there are a few workarounds for Mac [in this thread](https://github.com/arduino/Arduino/issues/1970).

gcampbell-msft, 2022-05-03:
This issue has been fixed in the latest release of this extension, which is available in the VS Code extension marketplace.

closed 2022-05-03 by gcampbell-msft

fussel132, 2022-05-03:
Thank you all for the answers, really helpful! I've just updated my extension and at least that massive log output is nearly gone. I now get a similar output like mentioned by @BelKed but two times in a row:

> [Starting] Verifying sketch 'sketch.ino'
Please see the build logs in output path: /Users/user/Library/CloudStorage/OneDrive-Persönlich/Dokumente/Arduino/sketch/.build
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
)
}
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
)
}
Konfiguration wird geladen...
Pakete werden initialisiert...
Boards werden vorbereitet...
Überprüfungsvorgang...
Wenig Arbeitsspeicher verfügbar, es können Stabilitätsprobleme auftreten.
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Cmd+Alt+I"
[Done] Verifying sketch 'sketch.ino'

Um maybe this could be resolved to only one time or never? As an end user I would prefer "never" to get an output similar to what you get on Windows when compiling a sketch (not these parameters just like "Loading configuration..." and so on). Or at least an option to toggle this, maybe moving it to the other log level "verbose"  so that you don't see these parameters on log level "info"

BelKed, 2022-05-09:
I'm almost certain that the content of `JVMOptions` wasn't listed there before in my case with the same settings. Probably, just some a problem with RegEx...

```
[Starting] Verifying sketch 'test.ino'
Please see the build logs in output path: /Users/user/Documents/Arduino/test/.vscode/Arduino
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
)
}
    "-Dapple.awt.application.name=Arduino",
    "-Dcom.apple.macos.use-file-dialog-packages=true",
    "-Dcom.apple.smallTabs=true",
    "-DAPP_DIR=$APP_ROOT/Contents/Java",
    "-Djava.ext.dirs=$JVM_RUNTIME/Contents/Home/lib/ext/:$JVM_RUNTIME/Contents/Home/jre/lib/ext/",
    "-Djava.net.preferIPv4Stack=true",
    "-Xdock:name=Arduino",
    "-Dcom.apple.mrj.application.apple.menu.about.name=Arduino",
    "-Dfile.encoding=UTF-8",
    "-Xms128M",
    "-Xmx512M",
    "-splash:$APP_ROOT/Contents/Java/lib/splash.png"
)
)
}
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
Sketch uses 743018 bytes (56%) of program storage space. Maximum is 1310720 bytes.
Global variables use 39400 bytes (12%) of dynamic memory, leaving 288280 bytes for local variables. Maximum is 327680 bytes.
IntelliSense configuration already up to date. To manually rebuild your IntelliSense configuration run "Cmd+Alt+I"
[Done] Verifying sketch 'test.ino'
```

gbretas, 2022-05-18:
The error persists in MacOs!

Frtrillo, 2022-07-12:
I've changed to use arduino-cli instead of the arduino ide for compilation, upload etc... and issue is solved ofc