microsoft/vscode-arduino #1495: 检测到 #include 错误。请更新 includePath。已为此翻译单元(D:\Projects\ArduinoProjects\arduino_vscode_nano\arduino_vscode_nano.ino)禁用波形曲线。

xiezhoubin, 2022-04-26:
When I use the rp2040 platform, this error will be displayed, but not for AVR, esp8266 and esp32.

[image]

xiezhoubin, 2022-04-26:
[image]
[image]

benmcmorran, 2022-04-29:
Thanks for the report @xiezhoubin. I can reproduce this issue with the Raspberry Pi Pico. I'm attaching English screenshots and some notes to assist future investigation.

I think this is happening because our command line parsing isn't correctly extracting the include paths. The error is complaining that `PeripheralNames.h` cannot be found, which (on my machine) in the `C:\Users\bemcmorr\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\3.0.1\cores\arduino\mbed\targets\TARGET_RASPBERRYPI\TARGET_RP2040`.

The compilation command line for the sketch looks roughly like this.

```
arm-none-eabi-g++ ... -iprefixC:\Users\bemcmorr\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\3.0.1\cores\arduino @C:\Users\bemcmorr\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\3.0.1\variants\RASPBERRY_PI_PICO/includes.txt arduino-investigation.ino.cpp -o arduino-investigation.ino.cpp.o
```

`includes.txt` has a line like this that contains the directory of interest.

```
-iwithprefixbefore/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040
```

This leverages [some fancy GCC options](https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html) to reuse parts of the include path. I'm guessing the command line parser doesn't understand these options and simply excludes the directory.

[image]

xiezhoubin, 2022-05-11:
So how to solve the problem of error reporting?

xiezhoubin, 2022-05-11:
Although the program can be compiled, this error report looks very uncomfortable, and also interferes with the error prompt of the code.

benmcmorran, 2022-06-06:
Unfortunately, there's isn't a great workaround right now. Fixing this problem requires changes to the command line parsing engine for IntelliSense that we haven't been able to prioritize. If you'd like to try to workaround the issue, you'll need to make sure that all the include paths defined in `%LOCALAPPDATA%\Arduino15\packages\arduino\hardware\mbed_rp2040\3.0.1\variants\RASPBERRY_PI_PICO\includes.txt` also exist in your c_cpp_properties.json file. However, there are a large number of include paths for the Pico so I don't think this is a great option.

xiezhoubin, 2022-06-20:
I hope this problem can be solved as soon as possible. Thank you very much!

jmdodd95682, 2022-07-22:
I'm also seeing this problem with Arduino RP2040 Connect.  It does not cause any failures in compile but it is very disturbing to see hundreds of errors flagged in my code.  I've disabled error squiggles just to make progress.  
I'm currently running VS Code 1.69.2 and Arduino IDE 1.8.19.  I've downloaded the latest RP2040 libraries.  Still the same problem.

Any estimate of when this could be fixed?

DawnMagnet, 2022-08-10:
I'm confused at these problems too.
[image]

VirensCn, 2022-11-28:
同样的问题，没法生成c_cpp_properties.json文件

`arduino.json`配置文件
```json
{
    "sketch": "SPIFFS_Test.ino",
    "port": "COM4",
    "output": "./build",
    "board": "esp32:esp32:esp32s3",
    "configuration": "PSRAM=enabled,FlashMode=dio,FlashSize=8M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default_8MB,CPUFreq=240,UploadSpeed=921600,DebugLevel=verbose,EraseFlash=none"
}

```

ECrecords, 2023-03-26:
This problem is still present, Im trying to use the Arduino Nano RP2040 Connect. Any updates?