microsoft/vscode-arduino #1497: Portables versions

HugoCar77, 2022-04-28:
Hi..!!

I use portables versions of vscode and arduino ide and for me both fine for now...but
I cant use 'Arduino Examples' directly from vscode if I change defaul sketchbook.path in for other in Arduino IDE

With defaul sketchbook.path run OK and open examples in a new window

[image]
[image]

Thanks!!

benmcmorran, 2022-04-29:
Hi @HugoCar77. To clarify, when you change the sketchbook path in the Arduino IDE, what happens to the "Arduino Examples" view? Does it not load at all, or are there some examples missing that you would expect to see there?

HugoCar77, 2022-04-30:
Hi...

Original path:
I can execute with double click
I can see everything

[image]

Personal path:
I can't run... double click and nothing happens
I also can't see "Personal Library Examples"

[image]

I have try even with a new windows install and an arduino and vscode install too

closed 2022-06-06 by github-actions[bot]