microsoft/vscode-arduino #1504: Serial Monitor not working on v0.4.12

NEXT-JP, 2022-05-12:
After updating VSCode-Arduino extension to `v0.4.12`, I am unable to see any messages printed over serial. I was still able to upload a sketch to my board and open the serial monitor, but no messages were displayed.

Serial port works as expected in the standalone Arduino app. I rewound to `v0.4.11` and serial communications were restored.

aleun, 2022-05-27:
Can you provide some extra info to help us reproduce the issue?

1. Operating system
2. What device are you using?
3. If possible, a link to a sample project demonstrating the problem

closed 2022-07-03 by github-actions[bot]

bassamjarad, 2022-11-19:
Hi,

This is indeed a problem. I've been playing with this arduino extension v0.4.12 and I've uploaded code to ESP32-CAM just fine, but I couldn't use the serial monitor to see the logs. Arduino IDE was working just fine and showing logs correctly.

I've just seen this issue and reverted back from v0.4.12 to v0.4.11 and suddenly, serial monitor started working as expected (COM11, baud: 115200)