microsoft/vscode-arduino #1506: forcedInclude should not apply to .cpp files (can I disable it entirely?)

tlokovic, 2022-05-20:
The Arduino build environment auto-includes Arduino.h from .ino files, but it does not do this for .h and .cpp files.

The VSCode Arduino extension tries to mimic this by adding a "forcedInclude" to the auto-generated c_cpp_properties.json, with a path to Arduino.h.  This forcedInclude applies to both .ino files and .cpp files, but it should only apply to .ino files.

The result is that Intellisense gets different compilation results on .cpp files than what you see when you run Verify.  In my case Intellisense is giving name collisions in .cpp files because it's auto-including Arduino.h when it shouldn't.

I'd be thrilled if I could just disable the forcedInclude entirely, but if I remove it from the workspace's c_cpp_properties.json, it just reappears moments later.  (I assume it's the Arduino extension that is doing that, but I'm unclear.)

Is there a way I can prevent this forcedInclude from constantly reappearing in c_cpp_properties.json?  (An even better fix would be for it to apply only to .ino files, but that sounds even harder.)

Thank you!

tlokovic, 2022-05-20:
I guess a workaround is to add

{
  "arduino.disableIntelliSenseAutoGen": true
}

to settings.json in the workspace.  This prevents c_cpp_properties.json from being constantly overwritten, so I can use that in the short term to make my removal of "forcedInclude" stick.  But if I end up switching devices I'll need to re-auto-generate, and I'll lose my tweaks.  So I'm still interested in a way of disabling the forcedInclude or making it smarter.

benmcmorran, 2022-06-06:
Thanks for the report @tlokovic! Unfortunately, I think disabling automatic generation as you described is currently the best workaround. We're considering more substantial changes to the IntelliSense processing in this extension that would bypass `c_cpp_properties.json` entirely and directly provide configuration information to the C/C++ extension. That should fix this (and other) IntelliSense issues, but it won't be available immediately. I'll leave this issue open for tracking progress.

alandsidel, 2022-09-20:
I just want to comment and my voice to getting better intellisense support.  I had no idea this setting existed and have been getting more than a little frustrated about having to go into my json file and make the same changes 50 times a day.  For me the changes are essentially to add browse paths so I can look up and jump to arduino library files and get autocompletion suggestions for libraries, and  of course to get rid of those annoying underlines on all the includes.

The workaround seems like it will work, though I haven't tried it yet, but it doesn't seem like it's going to be good long-term.

Some sort of option to merge a user-supplied json file with the autogenerated one could probably help alleviate my particular issue.