microsoft/vscode-arduino #1507: #include<avr/io.h> error c_cpp_properties.json automatic generation bug

al3x-huang, 2022-05-25:
VScode version: `1.67.2 (Universal) commit c3511e6`
Arduino extension version: `v0.4.12`

I am getting an include error `#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit` on `#include<avr/io.h>`.

This seems to be a pain point as there are a number of other users who reported this problem. I finally found a `c_cpp_properties.json` configuration that fixes this problem:
```
{
    "env": {
        "arduino_path": "/Applications/Arduino.app/Contents/Java",
        "arduino_avr_include_path": "${env:arduino_path}/hardware/arduino/avr",
        "arduino_avr_include2_path": "${env:arduino_path}/hardware/tools/avr/avr/include",
        "arduino_avr_compiler_path": "${env:arduino_path}/hardware/tools/avr/bin/avr-g++"
    },
    "configurations": [
        {
            "name": "Mac",
            "defines": [
                "ARDUINO=10810",
                "__AVR_ATmega328P__",
                "UBRRH"
            ],
            "includePath": [
                "${workspaceRoot}",
                "${env:arduino_avr_include_path}/**",
                "${env:arduino_avr_include2_path}/**"
            ],
            "forcedInclude": [
                "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/Arduino.h"
            ],
            "intelliSenseMode": "gcc-x64",
            "cStandard": "c11",
            "cppStandard": "c++11",
            "compilerPath": "${env:arduino_compiler_path} -std=gnu++11 -mmcu=atmega328p"
        }
    ],
    "version": 4
}
```
but every time, it kept getting overwritten by the Arduino extension and the automatically generated c_cpp_properties.json does not work. I discovered from  #438 that this is now a feature since 0.4.0. However, the c_cpp_properties.json generation logic seems buggy from my struggle with this #include error.

The current workaround is to disable the IntelliSense auto generation and use the above c_cpp_properties.json above. For anyone else having this issue here's a screen shot of the setting:
[image]

It would be great to see the automatic generation work though.

benmcmorran, 2022-06-06:
Thanks for the report and workaround @al3x-huang! I'm having trouble reproducing this issue. Can you provide some extra information?

- What operating system are you using? (I assume Mac from the configuration file)
- What Arduino device are you using? (Looks like maybe the Mega 2560, but I wanted to double check)
- Is the extension connected to the Arduino IDE or arduino-cli?

drzony, 2022-06-10:
I'm having a similar issue. It seems that the chip define is missing from the defines, in my case adding `__AVR_ATtiny2313__` to defines helps to solve the problem.
Running on Windows, using Arduino IDE with ATTiny2313

drzony, 2022-06-10:
After digging deeper it seems that cocopa does not pass `-mmcu` flag when requesting built-ins https://github.com/elektronikworkshop/cocopa/blob/ce125785574735b777e2a40bad0e85dd41580692/src/BuiltInInfoParserGcc.ts#L24

BelKed, 2022-06-11:
> I'm having a similar issue. It seems that the chip define is missing from the defines, in my case adding `__AVR_ATtiny2313__` to defines helps to solve the problem.

I think it has to do with this issue:
- https://github.com/microsoft/vscode-arduino/issues/1279

drzony, 2022-06-11:
Yes, that's the same issue