microsoft/vscode-arduino #1508: how to config compiler warning level

resos1, 2022-06-03:
Hello,
I'm looking for the setting to change the compiler warning level, equivalent config in preferences window of arduino ide (attached image)
[image]

I've already tried  #1392  but it doesn't worked for me, that kept adding -w parameter and ignoring the -Wall parameter.
I Prefer to set this setting to "Default", does it means that I just need to remove -w parameter, right? how can I do that?

benmcmorran, 2022-06-06:
Hi @resos1! #1392 is almost right, but you just need to set a different preference. Try adding this to your `arduino.json`. You can see all the preferences by viewing `%LOCALAPPDATA%\Arduino15\preferences.txt`.

```json
    "buildPreferences": [["compiler.warning_level", "default"]]
```

closed 2022-06-07 by resos1

Benik3, 2023-08-25:
Does this should still works?
I tried it and still got the same output. Maybe it doesn't work with the new CLI?
I tried to add into arduino.json:
```
"buildPreferences": [
        ["build.extra_flags", "-Wall"],
        ["compiler.cpp.extra_flags", "-Wall"],
        ["compiler.warning_level", "all"]
    ]
```

da-phil, 2024-06-04:
@Benik3 
> Does this should still works?

At least not for me, and it's clear why not, you already explained it:
The `buildPreferences` list is getting translated to `--build-property STRING`  arduino CLI arguments, and the actual config to change the warnings level is a separate CLI option `--warnings STRING`.

It would be really cool if vscode-arduino would expose the same config option in the main config, otherwise it's impossible to append this CLI option to the internal arduino CLI invocations.