microsoft/vscode-arduino #1509: cannot open source file "SoftwareSerial.h"

batonPiotr, 2022-06-03:
Cannot add standard arduino libraries to my project. Generated `c_cpp_properties.json` looks like this:

```
{
    "version": 4,
    "configurations": [
        {
            "name": "Arduino",
            "compilerPath": "/Users/USERNAME/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++",
            "compilerArgs": [
                "-std=gnu++11",
                "-fpermissive",
                "-fno-exceptions",
                "-ffunction-sections",
                "-fdata-sections",
                "-fno-threadsafe-statics",
                "-Wno-error=narrowing"
            ],
            "intelliSenseMode": "gcc-x64",
            "includePath": [
                "/Users/USERNAME/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino",
                "/Users/USERNAME/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/variants/standard",
                "/Users/USERNAME/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/lib/gcc/avr/7.3.0/include",
                "/Users/USERNAME/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/lib/gcc/avr/7.3.0/include-fixed",
                "/Users/USERNAME/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include"
            ],
            "forcedInclude": [
                "/Users/USERNAME/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino/Arduino.h"
            ],
            "cStandard": "c11",
            "cppStandard": "c++11",
            "defines": [
                "F_CPU=16000000L",
                "ARDUINO=10819",
                "ARDUINO_AVR_UNO",
                "ARDUINO_ARCH_AVR",
                "__DBL_MIN_EXP__=(-125)",
                "__HQ_FBIT__=15",
                "__cpp_attributes=200809",
                "__UINT_LEAST16_MAX__=0xffffU",
                "__ATOMIC_ACQUIRE=2",
                "__SFRACT_IBIT__=0",
                "__FLT_MIN__=1.17549435e-38F",
                "__GCC_IEC_559_COMPLEX=0",
                "__BUILTIN_AVR_SLEEP=1",
                "__BUILTIN_AVR_COUNTLSULLK=1",
                "__cpp_aggregate_nsdmi=201304",
                "__BUILTIN_AVR_COUNTLSULLR=1",
                "__UFRACT_MAX__=0XFFFFP-16UR",
                "__UINT_LEAST8_TYPE__=unsigned char",
                "__DQ_FBIT__=63",
                "__INTMAX_C(c)=c ## LL",
                "__ULFRACT_FBIT__=32",
                "__SACCUM_EPSILON__=0x1P-7HK",
                "__CHAR_BIT__=8",
                "__USQ_IBIT__=0",
                "__UINT8_MAX__=0xff",
                "__ACCUM_FBIT__=15",
                "__WINT_MAX__=0x7fff",
                "__FLT32_MIN_EXP__=(-125)",
                "__cpp_static_assert=200410",
                "__USFRACT_FBIT__=8",
                "__ORDER_LITTLE_ENDIAN__=1234",
                "__SIZE_MAX__=0xffffU",
                "__WCHAR_MAX__=0x7fff",
                "__LACCUM_IBIT__=32",
                "__DBL_DENORM_MIN__=double(1.40129846e-45L)",
                "__GCC_ATOMIC_CHAR_LOCK_FREE=1",
                "__GCC_IEC_559=0",
                "__FLT_EVAL_METHOD__=0",
[... 447 lines cut ...]
                "__GCC_ATOMIC_LLONG_LOCK_FREE=1",
                "__LDBL_DIG__=6",
                "__FLT_DECIMAL_DIG__=9",
                "__UINT_FAST16_MAX__=0xffffU",
                "__GCC_ATOMIC_SHORT_LOCK_FREE=1",
                "__BUILTIN_AVR_ABSHK=1",
                "__BUILTIN_AVR_FLASH_SEGMENT=1",
                "__INT_LEAST64_WIDTH__=64",
                "__ULLFRACT_MAX__=0XFFFFFFFFFFFFFFFFP-64ULLR",
                "__UINT_FAST8_TYPE__=unsigned char",
                "__USFRACT_EPSILON__=0x1P-8UHR",
                "__ULACCUM_FBIT__=32",
                "__QQ_IBIT__=0",
                "__cpp_init_captures=201304",
                "__ATOMIC_ACQ_REL=4",
                "__ATOMIC_RELEASE=3",
                "__BUILTIN_AVR_FMUL=1",
                "USBCON"
            ]
        }
    ]
}
```

When I `#include <SoftwareSerial.h>` in the arduino IDE it works just fine so it's not an issue with my arduino installation.

benmcmorran, 2022-06-06:
Hi @batonPiotr. I'm unable to reproduce this issue with a simple hello world project targeting the Arduino Uno. Can you provide some extra information to help us reproduce the issue?

- What operating system are you using?
- What version of the Arduino extension are you using?
- What Arduino device are you using?
- Is the extension connected to the Arduino IDE or arduino-cli?
- If possible, link to a sample project or provide source code that demonstrates the problem.

batonPiotr, 2022-06-07:
Thanks for the response. I've reinstalled the IDE, the extension, clicked "Verify" and it worked then. I cannot give you more details but the issue has been resolved.

For the backlog if it should happen again:

- What operating system are you using: macOS 12.4
- What version of the Arduino extension are you using? 0.4.12
- What Arduino device are you using? Arduino UNO R3
- Is the extension connected to the Arduino IDE or arduino-cli? IDE
- If possible, link to a sample project or provide source code that demonstrates the problem: It was a simple program with `SoftwareSerial.h` library

benmcmorran, 2022-06-07:
Glad you got it working, and thanks for providing the diagnostic info. I'll close this issue and tag it as IntelliSense in case we need it for future reference.

closed 2022-06-07 by benmcmorran