microsoft/vscode-arduino #1515: "More Info" button on library returns non-existent URL

zfields, 2022-06-10:
I found [this line](https://github.com/microsoft/vscode-arduino/blob/faf77358cf8ccc436d56a57e4fc3891c133b0e49/src/views/app/components/LibraryItemView.tsx#L115) in your code, but I can't tell where you are populating your library items from; specifically the "**More Info**" URI.

If you can tell me the base URI, then I can put another issue on Arduino and link back here.

gcampbell-msft, 2022-06-21:
@zfields I'm looking into where exactly the library items are populated from. In the meantime could you provide more information about your bug, like what library "more info" link returns a non-existent URL? 

Thanks!

gcampbell-msft, 2022-06-21:
@zfields I've done a deep dive and understand where the libraries are populated from, it'd a pretty complex so I won't go into it here. 
However, I can likely help you report this issue to Arduino or get it fixed once I know what link is broken. Thanks!

zfields, 2022-06-22:
If you search for `Blues Wireless Notecard` from the **Arduino: Library Manager** in VSCode, it should be the top result.

When you click **More Info**, then you are taken to `https://github.com/rayozzie/Notecard-Library`, instead of `https://github.com/blues/note-arduino`.

There is no mention of the invalid URL in the `library.properties` file, so I don't know how it's picking it up.

fearthecowboy, 2022-07-18:
I went to replicate this, and at first I was sure it wasn't going to do it for me, but sure enough it actually did.

Took me a couple minutes to see what's awry. 
It looks like the Arduino UI takes the 'more info' link from the latest version of the library in the feed, and the VSCode extension takes it from the first one in the feed. (which in this case is the *lowest* version)