microsoft/vscode-arduino #1517: c_cpp_preferences.json does not generate

boundmaidlea, 2022-06-16:
I am new to Arduino, and was annoyed with the Arduino IDE's poor ui customization, so i wanted to try this extension instead. I have read through countless different issues, issue comments, faqs, stackoverflow pages etc. for the last 3h and i am pretty dull by now so i'll probably forget some important details - sorry in advance.

The initial issue was that vsc gives me a wiggly line and a problem entry for each native Arduino function/variable.
I have found issues about that here, but they were closed with https://github.com/microsoft/vscode-arduino/pull/1141, so i looked in the .vscode folder and c_cpp_preferences.json was not there. 

I have tried the entry in the command palette and the keybind multiple times, but it doesn't do anything. I have tried to manually add that file but didn't get it to work, so i need the file that i should get from the extension.

The c_cpp_preferences.json file i am currently using makes the initial problem go away, but i still get an error about 
> cannot open source file "pins_arduino.h" (dependency of "C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Arduino.h")

While i am pretty confident i have identified the problem, as the file not getting generated, i do not have any clue about the cause of this problem.

Side note: I also came across https://github.com/microsoft/vscode-arduino/wiki/FAQ which just links to two removed sites and hasn't been changed since 2017 so i ignored it, even though it seemed to relate to my prolem at first... should probably be removed?

I don't really know what information could otherwise be helpful.

boundmaidlea, 2022-07-25:
well there is not really any more information i can give, especially as i haven't used vscode-arduino since since i am unable to use it.

Bludator, 2022-08-18:
I have the same issue with arduino-cli on win11:
<details>
<summary>My setup:</summary>

VSCode:
```
Version: 1.70.1 (system setup)
Commit: 6d9b74a70ca9c7733b29f0456fd8195364076dda
Date: 2022-08-10T06:08:33.642Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Windows_NT x64 10.0.22000
```
Extensions:
```
vsciot-vscode.vscode-arduino: v0.4.12
ms-vscode.cpptools: v1.11.5
```
Arduino-cli:
```
arduino-cli.exe  Version: 0.26.0 Commit: fc2ea723 Date: 2022-08-10T14:33:37Z
```
</details>
<details>
<summary>My config:</summary>

relevant options from settings.json:
```json
"arduino.path": "C:\\myPath\\arduino-cli_0.26.0_Windows_64bit",
"arduino.useArduinoCli": true,
"arduino.logLevel": "verbose"
```
</details>

---

I created new folder where I ran the VSCode, Inside, I create file `blink.ino` where I copy&paste code from: [arduino docs](https://docs.arduino.cc/built-in-examples/basics/Blink). When I try to upload it I am prompted to select board type etc. which I did and it results in `arduino.json`:
```json
{
    "board": "arduino:avr:uno",
    "sketch": "blink.ino",
    "port": "COM5",
    "programmer": "avrisp"
}
```
As consequesce:
- the C/C++ extension does not recognize build in functions and macros
- It fails with cryptic `Exit with code=1`:
[image]

supsm, 2022-11-01:
Same issue here, it works fine with arduino ide, just not with the cli. When I try to verify, upload, or do anything, this is what it outputs:
```
[Starting] Verifying sketch 'src/test.ino'
Please see the build logs in output path: C:\path\build
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'src/test.ino': Exit with code=1
```
The output folder is not created, and when I create it manually it is empty

arduino.json:
```json
{
    "port": "COM3",
    "configuration": "cpu=atmega328",
    "board": "arduino:avr:nano",
    "programmer": "avrisp",
    "sketch": "src/test.ino",
    "output": "build"
}
```
vs code:
```
1.72.2
d045a5eda657f4d7b676dedbfa7aab8207f8a075
x64
```
extension version: `v0.4.12`
arduino-cli: `0.28.0 Windows 64bit`
OS: windows 11

asclepiadae, 2022-11-30:
Very frustrating issue...

Because the Arduino IDE is broken with 2.x I manually adjusted the path and switched to CLI - but the IntelliSense configuration fails.

Really seems like this breaks Arduino within VSCode for the time being...

VirensCn, 2022-12-01:
me too，i use arduino-cli，all version is lastest

avsteele, 2023-10-10:
I also have this issue. 

- New install of Win11-Pro
- vs code installed (from download, not windows store, x64, 1.82.2).
- extension sinstalleda re Python, PyLand, C/C++, Github Copilot, and Arduino
- no version of Arduino IDE is installed, so I imagine just the `arduino-cli` that comes with the extension
- I added ACR boards using the board manager inside VS Code

Simple sketches compile for the Uno compile fine but no Intellisense config is generated. No c_cpp_propertes file either

```
[Starting] Analyzing sketch 'PC_PWM_Fan_controller.ino'
Please see the build logs in output path: C:\Users\avste\OneDrive\Desktop\tmp\PC_PWM_Fan_controller
Sketch uses 634 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
[Warning] Failed to generate IntelliSense configuration.
[Done] Analyzing sketch 'PC_PWM_Fan_controller.ino'
```

I think it may be related to a capitalization issue.

below, 'minimal sketch' refers to this

```
void setup(){}

void loop(){}
```

If I make a folder `test` and put in it a minimal sketch called `Test.ino` the sketch will fail to compile

```
[Starting] Verifying sketch 'Test.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
Error opening sketch: no valid sketch found in C:\Users\avste\OneDrive\Desktop\test: missing C:\Users\avste\OneDrive\Desktop\test\test.ino
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'Test.ino': Exit with code=1
```

This is expected file ino file has to match the folder (case sensitive). However If i now rename the file to `test.ino` the following happens:

```
[Starting] Verifying sketch 'Test.ino'
[Warning] Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.
Sketch uses 444 bytes (1%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
[Warning] Failed to generate IntelliSense configuration.
[Done] Verifying sketch 'Test.ino'
```

Now the sketch compiles but intellisense still fails. Not too the log file does not reflect the change from `Test.ino` to `test.ino`

If I delete everything and now instead make a new minimal sketch file `test.ino` then it both compiles and intellisense works.

Khalid-Hersafril-Flowcopter, 2024-06-14:
Hi, 

I have the issue and I am currently using Linux. Here's my machine information

```
uname -a
>> Linux KhalidOwlWalid 6.5.0-25-generic #25~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Feb 20 16:09:15 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```

I think it is just the fact that the issue stems from the way this vscode extension parses the file information or perhaps Arduino IDE in general.

For example, if I open an example file in Linux, the file will be stored in 

```
/home/<user>/Arduino/generated_examples
```

If this is your first time opening the example, if you try to verify the code, it will work out fine, but if you close the example, and then re-open the same example using the command pallete, this will generate a 2nd file with the following naming: 

<example_file>**_x** where **x** is the number of times you have open the example

Example: If I have open an example called `CAN_send` thrice, this will generate a file which is called `CAN_send_3`. Note that the `.ino` file inside the example file is not labelled as `CAN_send_3.ino` but instead `CAN_send.ino`. When you try to verify the code, it will fail as the `.ino` file needs to be the same name as the `.ino` file. A simple fix would be renaming the `.ino` file to be the same name as the folder.

Here's a video demonstrating the issue above:

[image]

This is quite frustating, since it has been a known bug for quite a long time. Hopefully, this will help anyone else reproduce the bug. I'd assume this should be similar to Windows since @avsteele mentioned about his `Test` folder and it seems to be of similar case.