microsoft/vscode-arduino #1522: Arduino Extension does not respect settings in Dev Container

zfields, 2022-06-26:
When developing Arduino in a dev container, I discovered that the Arduino Extension requires settings to be set on the **User** tab, and does not respect settings in the **Remote [Dev Container: ...** tab.

This is a serious problem, because the container frequently has different paths for tools than my host machine (assuming they are installed at all). This makes switching between dev container development and native host development a pain, because I have to break one to use the other.

`devcontainer.json`:

```json
// For format details, see https://aka.ms/devcontainer.json. For config options, see the README at:
// https://github.com/microsoft/vscode-dev-containers/tree/v0.205.2/containers/docker-existing-dockerfile
{
    "name": "ESP32 Development Environment Dockerfile",

    // Sets the run context to one level up instead of the .devcontainer folder.
    "context": "..",

    // Update the 'dockerFile' property if you aren't using the standard 'Dockerfile' filename.
    "dockerFile": "../Dockerfile",

    // Set *default* container specific settings.json values on container create.
    "settings": {
        "arduino.analyzeOnOpen": false,
        "arduino.analyzeOnSettingChange": false,
        "arduino.logLevel": "verbose",
        "arduino.path": "/usr/local/bin/arduino-cli",
        "arduino.useArduinoCli": true,
        "C_Cpp.autoAddFileAssociations": false,
        "C_Cpp.default.compilerPath": "${config:esp_toolchain_path}/xtensa-esp32-elf-gcc",
        "C_Cpp.default.includePath": [
            "${default}",
            "${workspaceFolder}/build/config",
            "${config:esp_idf_path}/components/bootloader_support/include",
            "${config:esp_idf_path}/components/bootloader_support/bootloader_flash/include",
            "${config:esp_idf_path}/components/efuse/include",
            "${config:esp_idf_path}/components/efuse/esp32/include/",
            "${config:esp_idf_path}/components/esp_common/include",
            "${config:esp_idf_path}/components/esp_hw_support/include",
            "${config:esp_idf_path}/components/esp_hw_support/include/soc",
            "${config:esp_idf_path}/components/esp_rom/include",
            "${config:esp_idf_path}/components/esp_system/include",
            "${config:esp_idf_path}/components/esp_system/port/include",
            "${config:esp_idf_path}/components/esp_system/port/include/private",
            "${config:esp_idf_path}/components/hal/include",
            "${config:esp_idf_path}/components/hal/esp32/include",
            "${config:esp_idf_path}/components/hal/platform_port/include",
            "${config:esp_idf_path}/components/soc/include",
            "${config:esp_idf_path}/components/soc/esp32/include",
            "${config:esp_idf_path}/components/soc/esp32/include/soc",
            "${config:esp_idf_path}/components/spi_flash/include",
            "${config:esp_idf_path}/components/spi_flash/sim/stubs/log/include",
            "${config:esp_idf_path}/components/xtensa/include",
            "${config:esp_idf_path}/components/xtensa/esp32/include",
        ],
        "terminal.integrated.defaultProfile.linux": "bash",
        "terminal.integrated.env.linux": {
            "PATH": "${config:esp_dev_env_path}:${env:PATH}"
        }
    },

    // Add the IDs of extensions you want installed when the container is created.
    "extensions": [
		"ms-vscode.cpptools",
		"twxs.cmake",
		"vsciot-vscode.vscode-arduino"
	],

    // Use 'forwardPorts' to make a list of ports inside the container available locally.
    // "forwardPorts": [],

    // Uncomment the next line to run commands after the container is created - for example installing curl.
    //"postCreateCommand": "apt-get update && apt-get install -y curl",
[... 155 lines cut ...]
    curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=${BINDIR} sh -s ${ARDUINO_CLI_VERSION} \
 && mkdir -p /etc/bash_completion.d/ \
 && arduino-cli completion bash > /etc/bash_completion.d/arduino-cli.sh \
 && echo >> /etc/bash.bashrc \
 && echo \"for bcfile in /etc/bash_completion.d/* ; do\" >> /etc/bash.bashrc \
 && echo \"    [ -f \\\"${ECHO_BC_FILE}\\\" ] && . \\\"${ECHO_BC_FILE}\\\"\" >> /etc/bash.bashrc \
 && echo \"done\" >> /etc/bash.bashrc \
 && echo \"if [ -f /etc/bash_completion ]; then\" >> /etc/bash.bashrc \
 && echo \"    . /etc/bash_completion\" >> /etc/bash.bashrc \
 && echo \"fi\" >> /etc/bash.bashrc \
"]

# Configure Arduino CLI as non-root user
WORKDIR /home/${USER}/
USER ${USER}

# Configure Arduino CLI
RUN ["dash", "-c", "\
    arduino-cli config init \
 && arduino-cli config add board_manager.additional_urls \
     https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json \
 && arduino-cli core update-index \
 && arduino-cli core install esp32:esp32 \
"]
```

zfields, 2022-06-26:
It turns out I was confused by the settings names.

To get the Arduino CLI to run in a container, I needed the following settings listed in `devcontainer.json`

```json
"settings": {
        "arduino.commandPath": "arduino-cli",
        "arduino.path": "/usr/local/bin",
        "arduino.useArduinoCli": true,
        ...
```

The setting named `arduino.commandPath` could probably be named `arduino.commandExecutable`. That would have clarified everything for me.

> _**NOTE:** I'm leaving this open for feedback/visibility, but any member of the Arduino Extension repo should feel free to close it._

gcampbell-msft, 2022-07-12:
@zfields Thanks for the comments and working through the issue! I will go ahead and close the issue since the problem was resolved, however, thanks for the feedback regarding the settings name!

closed 2022-07-12 by gcampbell-msft