microsoft/vscode-arduino #1523: cannot find arduino IDE, i'm using WSL with arduino inside WSL

Mettikodeva-Nara, 2022-07-13:
where should I specify the path of the arduino inside the WSL, in User or Remote[WSL:Ubuntu]?
[image]
or I can only use arduino inside windows?
is there any way to compile with arduino inside wsl from this extension?

fearthecowboy, 2022-07-18:
Do you have the linux version of arduino installed inside the WSL instance?

Mettikodeva, 2022-07-19:
yes, i've installed the arduino and set the path inside the vscode
[image]

Mettikodeva, 2022-07-19:
yes i've installed the arduino and i've set the path too
[image]

gcampbell-msft, 2022-08-08:
@Mettikodeva Could you show us what error you are seeing? It might help us diagnose where the issue is arising. Thanks!

David00, 2022-09-11:
I'm getting this issue too. 

### Settings.json:

```
$ cat ~/.vscode-server/data/Machine/settings.json | grep "arduino"
    "arduino.path": "/home/david/.local/bin/arduino-cli"
```

### Testing the `arduino-cli` package at the path above:

```
$ /home/david/.local/bin/arduino-cli version
arduino-cli  Version: 0.27.1 Commit: a900cfb2 Date: 2022-09-06T16:44:27Z
```

When I choose Arduino: Initialize from the extension actions, I get the message:

> Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change.

I have restarted VS Code.

VS Code Version: 1.71.0
Windows 11 x64
WSL Details:
```
$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.4 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.4 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

Let me know if you need more information.

borhanreo, 2022-09-16:
### **_My arduino Version_**
[image]

### **_Install location of arduino version 2.0.0_**
[image]

### **_Still i did not resolve the issue_**
[image]

Could you please give me help, what i have miss

arminrosu, 2022-10-14:
@borhanreo Arduino IDE 2.0 is not supported (see https://github.com/microsoft/vscode-arduino/issues/1477). Use the [CLI](https://github.com/microsoft/vscode-arduino#arduino-cli) to get the extension working.

closed 2022-11-20 by github-actions[bot]