microsoft/vscode-arduino #1526: Serial console is very slow

Rofl92, 2022-07-23:
Hello.
VSCode `v1.69.2`
vscode-arduino `v0.4.12`

When i connect to Serial via Arduino-vscode, i have very slow output. (1st video)

https://user-images.githubusercontent.com/25763319/180605333-2c083c77-a3d4-41f2-bad5-ebd6d197d364.mov

But if i connect to Serial via Terminal output speed is very good.  (2nd video)

https://user-images.githubusercontent.com/25763319/180605281-d942b26a-f758-4c07-a9b3-0292efe7638e.mov

How can i solve this problem?

BelKed, 2022-07-23:
It's a known problem, and it seems to be already fixed but not yet released...
- https://github.com/microsoft/vscode-serial-monitor/issues/23
- https://github.com/microsoft/vscode-arduino/issues/1396

gcampbell-msft, 2022-08-08:
@BelKed Thanks for updating @Rofl92, you are correct! The issue is fixed and is released in the vscode-serial-monitor extension. I'll close this issue, but please open another if the issue persists. Thanks!

closed 2022-08-08 by gcampbell-msft