microsoft/vscode-arduino #1528: un-useful error message about output path and mystery README file

gitcnd, 2022-08-26:
I see this:

Output path is not specified. Unable to reuse previously compiled files. Build will be slower. See README.

This post: https://arduino.stackexchange.com/questions/45347/warning-when-verifying-sketch-with-vs-code

suggests:  in arduino.json settings file add: "output": "../ArduinoOutput"

... which appears to be more directly to the point - if that's the correct answer, might I suggest fixing the error to be useful (and any other vague ones that might also be lurking where this came from, if possible/easy?)

gitcnd, 2022-08-26:
suggestion... since something is already creating a per-project folder ./.vscode it might be a good idea to have a default "output" folder of ./.vscode/.build to avoid that nuisance entirely?

Using a single global build folder looks like a not-so-good idea, since the files it puts there have no parent-folder information included in them, so any projects using files and libraries of the same name may clobber each other?

benmcmorran, 2022-09-02:
Thanks for the report. How are you currently setting up your Arduino project in VS Code? #1425 should have done what you suggest and set "build" as the default build directory for projects initialized through the Arduino: Initialize command, but it won't change the behavior for existing projects. We can consider applying the default universally, but I want to be careful we don't cause unexpected behavior relative to the Arduino IDE. Would you expect the VS Code extension to ask if you'd like to use the default directory, or simply use it silently?

stsvilik, 2022-09-08:
Is it normal that output directory gets cleaned up on every "verify"?

gcampbell-msft, 2022-09-09:
@stsvilik Yes I believe it is, however if this isn't desired please feel free to post a separate issue, but I do believe that it's expected as of now.

closed 2022-10-16 by github-actions[bot]