microsoft/vscode-arduino #1529: Intellisense autogen fails to include definitions for PORTA

BjornMoren, 2022-08-28:
It seems like the intellisense autogen that inserts items into the defines list in c_cpp_properties.json is not working properly.

My project compiles fine, but I get intellisense warnings in my Ardunio code that PORTA is undefined.

If I manually add "__AVR_ATmega328P__" to the defines list in c_cpp_properties.json then intellisense works fine.

benmcmorran, 2022-09-02:
Thanks for the report. To help us reproduce the issue, can you provide a sample `.ino` file that reproduces the issue and tell us which board you are targeting? You can find the board in the `.vscode/arduino.json` file.

BjornMoren, 2022-09-03:
From arduino.json: 
"board": "arduino:avr:uno"

Project.ino has no issues, uses the following includes:

#include <Arduino.h>
#include "ScreenLCD.h"
#include "DateTime.h"

#if ENABLE_SENSORS == 1
#include <Wire.h>
#endif

#if ENABLE_SD_CARD == 1
#include <SPI.h>
#include "SdFat.h"
#endif

The ScreenLCD.cpp has the intellisense issues, here are the relevant parts:

#include "ScreenLCD.h"
#ifdef __AVR__
  #include <avr/pgmspace.h>
#elif defined(ESP8266) || defined(ESP32)
  #include <pgmspace.h>
#endif

void ScreenLCD::_writeCommandData8(uint8_t reg, uint8_t data)
{
	PORTA = 0; <-- Intellisense complains about PORTA
}

Here is the beginning of the defines section of the c_cpp_properties.json file:
            "defines": [
                "F_CPU=16000000L",
                "ARDUINO=10816",
                "ARDUINO_AVR_UNO",
                "ARDUINO_ARCH_AVR",
                "__AVR_ATmega328P__",        <-- This one I have to manually add myself, and then turn off autogen
                "__DBL_MIN_EXP__=(-125)",

al3x-huang, 2022-11-21:
Still running into this issue.

StefanKellerAC, 2022-12-10:
maybe it is related to https://github.com/microsoft/vscode-arduino/issues/1354#issuecomment-1345237735 in the Pico core.