microsoft/vscode-arduino #1530: Not work with the IBM Output Colorizer plugin.

nmpassthf, 2022-08-31:
Does not work with the Output Colorizer plugin.
When I enable this plugin, Output Colorizer stops working and cannot colorize any text in the output.

The Output Colorizer plugin: [link](https://github.com/IBM-Cloud/vscode-log-output-colorizer)

BelKed, 2022-08-31:
Duplicate of https://github.com/microsoft/vscode-arduino/issues/1447

benmcmorran, 2022-09-02:
As mentioned in the linked issue, this is a [limitation of VS Code itself](https://github.com/microsoft/vscode/issues/11005) and [any other extension that provides a grammar for the output pane will likely hit the same issue](https://github.com/IBM-Cloud/vscode-log-output-colorizer/issues/21#issuecomment-392047772). Feel free to discuss on the VS Code issue tracker.

closed 2022-09-02 by benmcmorran