microsoft/vscode-arduino #1537: Issue with Arduino extension installation in VS Code

Jcat2b, 2022-09-25:
Hello,

I have an issue when I try to install the Arduino  extension in VS Code. After the installation I launch "Arduino: Board Manager" but I have this issue: "command 'arduino.showBoardManager' not found".
It seems to be a wrong Arduino Path issue but I can't solve it (I put some screeshots to help to undestand the issue).

Do you know how to solve this issue?

Thank you.
[image]
[image]

aliberkarad, 2022-09-29:
Visual Studio Code Arduino Plugin not work with Arduino 2.0 , you should with Arduino 1.8.x

Jcat2b, 2022-09-29:
> Visual Studio Code Arduino Plugin not work with Arduino 2.0 , you should with Arduino 1.8.x

Hello,

Thank you for your answer. I uninstall Arduino 2.0, then I install Arduino 1.8.19, I changed the parameters in Visual Studio Code (path for Arduino and "arduino.exe" file name).

But I still have the issue when I try to launch "Arduino: Board Manager", do you see another reason for this problem?

aliberkarad, 2022-09-29:
Hello , 
Sorry But I don't know any other reason. I uninstalled Arduino 2.0 when I got the error. I downloaded Arduino 1.8.9 instead of it. Then I did re-install Arduino plugin.And my issue solved.

Jcat2b, 2022-09-29:
> the

I have version 1.71.2 for VS Code and v0.4.12 for Arduino App on VS Code.
Is it the same for you?

aliberkarad, 2022-09-29:
Yes , Its same version as mine

Jcat2b, 2022-09-29:
> Yes , Its same version as mine

Ok thank you, I will try to unistall all and install them again.

butcher51, 2022-10-27:
Yeah, this bug is there, it seems it can't handle the space in the folder/path.

benmcmorran, 2023-02-16:
Thanks for the feedback. [Only the legacy IDE is currently officially supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

closed 2023-03-26 by github-actions[bot]