microsoft/vscode-arduino #1538: Cannot find Arduino Path Mac

T0mHutter, 2022-09-30:
After I installed the IDE and the extension i cannot make out why its not working. 
I have tried to reinstall both the ide and vscode (uninstalled the extension before that), 
i tried mulitple locations for the extension (downloads/ applications) 

I dont use the Cli and i have the command path set to 'Contents/MacOS/Arduino'. 
I've also tried using it without the command path

arminrosu, 2022-10-13:
Are you also using Arduino IDE 2.0? Looks like it's not currently supported: https://github.com/microsoft/vscode-arduino/issues/1477

I've set up the extension to use the arduino-cli instead of the IDE.

Just `brew install arduino-cli`, then follow this guide: https://medium.com/@thomas.kilmar/arduino-cli-with-visual-studio-code-on-macos-d2ad32ff0276. Replace the _Arduino: Path_ value there with your path output by `which arduino-cli`

T0mHutter, 2022-10-14:
Thx that worked for me :) 
Just have to edit the include Path now

T0mHutter, 2022-10-14:
Because I'm getting errors at for example serial etc.

arminrosu, 2022-10-14:
Do you mean the typings? You need to run "Verify" on the project. That will create a correct .vscode/c_cpp_properties.json for you and typings will work.

T0mHutter, 2022-10-14:
Oh that was my mistake

T0mHutter, 2022-10-14:
I will close this issue now

closed 2022-10-14 by T0mHutter