microsoft/vscode-arduino #1540: Bug

juliusRomanzeira, 2022-10-12:
I just tried a lot of times use ide arduino with vscode. I'll attach the error here:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

arminrosu, 2022-10-13:
Arduino IDE 2.0? See https://github.com/microsoft/vscode-arduino/issues/1477

hlovdal, 2022-12-26:
Could you update the title of this issue to something more descriptive than just "Bug"?

Imagine what nightmare it would be for the maintainers if all the issues listed at https://github.com/microsoft/vscode-arduino/issues just had titles "Bug" or "Feature request".

hlovdal, 2022-12-26:
#1548

benmcmorran, 2023-02-16:
Thanks for the feedback. [Only the legacy IDE is currently officially supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

closed 2023-03-26 by github-actions[bot]