microsoft/vscode-arduino #1541: [Warning] Failed to generate IntelliSense configuration.

lukasgabriel, 2022-10-14:
When trying to generate the IntelliSense configuration in `c_cpp_properties.json`, I get this error:
[image]

This makes the extension pretty much unusable since no IntelliSense info is there - might as well use the Arduino IDE.
[image]

The sketch name is the same name as the folder it is in.
The problem also appears in a fresh project.
I have tried reinstalling the Arduino and C/C++ extensions.
I've also uninstalled Visual Studio, VS Build Tools and several other things that could have interfered (as described in other issues with this problem).
I've tried everything I could find and think of for the past 2 hours, no luck. Any help would be greatly appreciated.

System info:
Windows 11 Pro for Workstation 22H2 22623.746
VSCode Version 1.72.2
Arduino and C/C++ Extensions on their respective newest versions.
Board: Raspberry Pi Pico 

Here's my `arduino.json`:
```
{
    "arduino.path": "C:/ProgramData/chocolatey/bin",
    "sketch": "client\\client.ino",
    "configuration": "flash=2097152_0,freq=133,opt=Small,rtti=Disabled,stackprotect=Disabled,exceptions=Disabled,dbgport=Disabled,dbglvl=None,usbstack=picosdk,ipstack=ipv4only",
    "board": "rp2040:rp2040:rpipico",
    "port": "COM8",
    "output": "output"
}
```

And here's my workspace `settings.json`:
```
{
    "cmake.configureOnOpen": false,
    "arduino.path": "C:\\ProgramData\\chocolatey\\bin",
    "arduino.useArduinoCli": true,
}
```

This is probably related to #1373 #1489 #1514 #1517

olifink, 2022-11-16:
I had the same problem, was foolishly trying to use new software like `arduino-cli` and alternatively the new Arduino IDE 2.x. But both didn't work, however by installing Arduino IDE 1.8 and removing the arduino settings both from the project and user settings before trying again resulted in an auto generated `c_cpp_properties.json` for the Arduino config that handles IntelliSense correctly

DoganM95, 2022-11-30:
i fought with this nonsense for 2 days now. am tired of this extremely buggy and poorly maintained.
Switched to PlatformIO now, a bit unconvenient but it works at least...

lukasgabriel, 2022-12-04:
Sadly, looking at all the unadressed issues, this repo (and extension) seems to be in a pretty poor state and not really usable right now.
I think it really deserves some attention since VSCode these days can do 99% of things, sucks having to leave that workflow and familiar environment for Arduino things.

Scoutzknifez, 2022-12-05:
> I had the same problem, was foolishly trying to use new software like `arduino-cli` and alternatively the new Arduino IDE 2.x. But both didn't work, however by installing Arduino IDE 1.8 and [...]

This worked for me as well.
Make sure your VSCode settings for Arduino.path are empty.  I deleted the c_cpp_properties.json and it regenerated it completely fine.
Took a while to come to this solution though, I do wish it was more straight forward.

jamesrwaugh, 2022-12-27:
I found I needed to actually open the Arduino IDE, verify some code, and install a library before the extension worked. I'm assuming this places compilation C/C++ files on the system that the extension uses.

On Ubuntu:
1. Started with a fresh arduino.json / settings.json
2. Install Ardunio IDE 1.8.x using the provided `install.sh`
3. Plug in my Arduino
4. Upload blank sketch, add any libraries I'm using
5. Set extension paths accordingly, for me these were left blank as I used default `install.sh` options
6. Verify / generate now works.

javos65, 2023-01-13:
I went through this VS-Hell as well on Windows , The point for Arduino compiler is that :

Arduino IDE 1.xx is supported , as it supports cli  (command line interface) calls that VS use to operate.
Arduino IDE 2.00 is NOT supported, as it does not support cli-calls
Arduino PRO is NOT supported - idem
Arduino-CLI is supported, this is the ' under the hood' version of IDE2.0 / PRO

I'v setup Arduino-CLI in VS, specifying the settings.json for compiler specifics:
{
    "arduino.enableUSBDetection": true,
    "arduino.path": "C:\\Program Files\\arduino-cli_0.29.0",
    "arduino.commandPath":"arduino-cli.exe",
    "arduino.useArduinoCli": true,
}

ALSO, check your arduino-cli.yaml  init-file to set the arduino directories correct for library recognition.
it in the  your AppData\Local\Arduino15 directory  : 

directories:
  data: C:\Users\<>\AppData\Local\Arduino15
  downloads: C:\Users\<>\AppData\Local\Arduino15\staging
  user: C:\Users\<>\OneDrive\Documents\Arduino

Last but not least: if your have custom libraries in use, CLI does not recognize them as it does not scan your library directory, it looks it up in an index-file. I had to re-install custom libs it using github (cli-command line install)

In VS the compiler environment and the C++ code-check environment are not the same, so check the C++ browse path to have code support.

i'v not been able to solve the intellisense support.

just my 50 cents.

laurensV, 2023-01-13:
Same problem here, Arduino IDE 2.x (officialy released now) does not work, and when trying to use the arduino-cli (either the one that is already included in Arduino IDE 2.x or a freshly downloaded arduino-cli.exe file) have the problen that the libraries/dependencies are not (things like cannot open source file "Wire.h") and when trying to verify the project, I get the warning: Failed to generate IntelliSense configuration. Extension is not usable in this state..

MKme, 2023-01-18:
Same problems as above here.  Have yet to get mine to work.  Intellisense and Lib errors.  The Lib errors I resolved with Arduino-CLI but then intellisense ruins the day.  Wasted a week on this

anzbert, 2023-01-31:
It works fine on macos, just with the CLI and Arduino 2.0 installed.

On windows on the in other hand, I need the old arduino 1.8.x installed for the extension to work properly.

benmcmorran, 2023-02-17:
Can you try again with the recently released version 0.5.0 of the extension? It includes #1591, a fix to the extension's handling of Arduino CLI for generating IntelliSense configurations.

marsfan, 2023-02-17:
@benmcmorran I'm having issues with 0.5.0. The command `Arduino: Verify` only works every third time. Seems that it likes to wipe and re-generate the `compile_commands.json` file. Could this be related or should I open a separate bug? 

End of Log Output After First Error: 
```log
Error during build: unable to read contents of the source item: open C:\Users\Gabe Roper\Documents\Arduino\simple_esp32_audio\.build\compile_commands.json: The system cannot find the file specified.

Used platform Version Path                                                                           
esp32:esp32   2.0.3   C:\Users\Gabe Roper\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'simple_esp32_audio.ino': Exit with code=1
```

End of Log Output After Second Error:
```log
Error during build: unable to read contents of the source item: open C:\Users\Gabe Roper\Documents\Arduino\simple_esp32_audio\.build\compile_commands.json: The system cannot find the file specified.
[Warning] Failed to generate IntelliSense configuration.
[Error] Verifying sketch 'simple_esp32_audio.ino': Exit with code=1
```

Contents of `compile_commands.json` after first error: 
```json
[]
```

Contents of `arduino.json`
```json
{
    "configuration": "PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none",
    "board": "esp32:esp32:esp32",
    "sketch": "simple_esp32_audio.ino",
    "output": "./.build"
}
```

benmcmorran, 2023-02-17:
Thanks for checking. Based on the error messages it looks likely that this is the same issue as https://github.com/arduino/arduino-cli/issues/1646. As a workaround, you can try changing the `output` setting to a path outside the sketch directory.

marsfan, 2023-02-18:
@benmcmorran Moving the output directory to be in a path outside of the sketch directory seems to have fixed it for me. Not sure if that solves the problem for anyone else though.

toamz, 2023-02-18:
Changing the output directory to be outside of the sketch directory fixed it for me too.

reflectingme, 2023-03-04:
> It works fine on macos, just with the CLI and Arduino 2.0 installed.
> 
> On windows on the in other hand, I need the old arduino 1.8.x installed for the extension to work properly.

I am on MacOS and cannot get this extension to work, it is so frustrating.  Regular errors experienced are unable to locate include files, and failed to generate Intellisense configuration.

I really don't like the ArduinoIDE and would much prefer VSCode.  I load a file in ArduinoIDE and it compiles and uploads perfectly, load the very same file in the very same sketch folder using VSCode and it fails to find the include files and fails to generate Intellisense configuration.  Pulling out what little hair I have left, lol.

anzbert, 2023-03-05:
hey @reflectingme . I am not sure why its not working for you. Since the latest update includes arduino-cli, it should be even easier to set up.

I am using the following most important settings on my mac with M1 chip:
- Arduino: Command Path and Arduino: Path: Both left empty
- Arduino: Use Arduino CLI: Ticked!

That seems to do the trick. Otherwise, I do have Arduino IDE 2.0 installed on my Mac. Not sure if that makes a difference.

Good Luck :)

reflectingme, 2023-03-05:
Thanks @anzbert - it is causing me to pull out the little remaining hair I have left :)

I am not on M1 but am using the same setup, both parameters empty and use Arduino cli ticked, and I have Arduino IDE 2.0 running, but I still get these darned intellisense and include errors. I run a file in the Arduino IDE and it works fine, and when I try to load and compile the very same file using VScode it does not even compile. So very annoying as I’m pretty much forced to use the IDE when I’d much rather use VScode. 

Thanks though for your reply.

anzbert, 2023-03-05:
Have you tried installed the old Arduino pre2.0 software  as well, seeing
if that makes a difference?!

I am just stabbing in the dark here as well?!

closed 2023-04-11 by github-actions[bot]

couillonnade, 2023-05-03:
For me, the solution came after I opened the .ino file in Arduino IDE and it complained that the file must be in a folder with the same name, proposed to create the folder and move the file. After that, I was able to use vscode. Conclusion, .ino file cannot be at the root of the workspace.

tingjhenjiang, 2023-06-13:
In my case, after I change the "name" part from "Win32" to "Arduino" in "c_cpp_properties.json", everything would go fine. VSCode would start to rebuilt IntelliSense configuration.

lobodpav, 2023-10-14:
For me, VSCode always reports errors in my `.ino` file, complaining that
```
cannot open source file "mbed_toolchain.h" (dependency of "WiFi.h")C/C++(1696)
cannot open source file "netsocket/NetworkInterface.h" (dependency of "thingProperties.h")C/C++(1696)
```

If I apply the Quick Fixes offered that update the `includePath`, it starts complaining again, this time
```
cannot open source file "EthernetInterface.h" (dependency of "thingProperties.h")C/C++(1696)
```

The problem goes on no matter how many times I apply the quick fixes, and after verifying the code or rebuilding IntelliSense, all these quick fixes get reverted.

aklein24, 2024-01-06:
I tried so many of the things here without any luck. I was looking again at [tingjhenjiang](https://github.com/tingjhenjiang)'s comment, and then I noticed in the bottom right of the screen that I had the Win32 C++ configuration selected. Changed that to Arduino and my problem went away!

Clicking this in the bottom right corner:
[image]

Will bring up the command palette, and then changing this to Arduino finally fixed it:
[image]
[image]

This issue has been annoying to me for _years_, but I just ignored it because everything worked OK. Thanks [tingjhenjiang](https://github.com/tingjhenjiang) for putting me on the right path!

kameo4242, 2024-01-15:
Rahhhhh, this thing drove me mad for days...
It's painful but can actually work, with intellisense! The real problem is I don't know what exactly made it work :/
(but I've a screen capture to prove it's real). Good news is: I've a working config I can share ;)

**Current setup:** 
- Windows 11 (22h2)
- Arduino IDE 2.2.1
- Arduino CLI 0.35.0
- Visual Studio Code 1.85.1 (system, not user version)
- Arduino extension Release version 0.6.230727001
- C/C++ extension v1.18.5
- Cmake v0.0.17
- Cmake tools v1.16.32
- UE Intellisense Fixes v3.1.4 (Can help fix some weirdnesses it seems)

What I think could have played a role:
- Full fresh install of vscode with wiping of all data in Appdata\roaming\code
- My project are stored in a directory without any space in their name to avoid Windows vs Posix issues
- I renamed my profile from win32 to arduino as proposed above
- I installed the beta version of Arduino extension
- I edited hundreds of times my config files with all kind of weird attempts until these one worked
- They know I'm a stubborn donkey

c_cpp_properties.json would be long to share but it was properly generated by a Ctrl+alt+I (regen intellisense config). 

**.vscode\settings.json**
_{
    "arduino.clearOutputOnBuild": true,
    "arduino.useArduinoCli": true,
    "arduino.path": "F:\\code\\compiler\\arduino-cli.exe",
    "arduino.commandPath": "arduino-cli.exe",
    "arduino.enableUSBDetection": true,
    "workbench.colorTheme": "Default Dark Modern",
    "cmake.configureOnOpen": true,
    "cmake.options.statusBarVisibility": "icon",
    "C_Cpp.default.compilerPath": "F:\\code\\compiler\\arduino-cli.exe"
}_

**c:\user\me\AppData\Roaming\code\User\settings.json**
_{
    "arduino.clearOutputOnBuild": true,
    "arduino.useArduinoCli": true,
    "arduino.path": "F:\\code\\compiler\\arduino-cli.exe",
    "arduino.commandPath": "arduino-cli.exe",
    "arduino.enableUSBDetection": true,
    "workbench.colorTheme": "Default Dark Modern",
    "cmake.configureOnOpen": true,
    "cmake.options.statusBarVisibility": "icon",
    "cmake.showOptionsMovedNotification": false,
    "C_Cpp.default.compilerPath": "F:\\code\\compiler\\arduino-cli.exe",
    "UEIntellisenseFixes.enableFixes": true,
    "UEIntellisenseFixes.enableOptionalFixes": true,
    "UEIntellisenseFixes.compiler.path": "F:\\code\\compiler\\arduino-cli.exe",
    "security.workspace.trust.untrustedFiles": "open"
}_

**.vscode\arduino.json**
_{
    "configuration": "JTAGAdapter=default,PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=none,EraseFlash=none",
    "board": "esp32:esp32:esp32s3",
    "programmer": "esptool",
    "sketch": "AIBOX\\AIBOX.ino",
    "output": "f:\\code\\AIBOX"
}_

**c:\user\me\AppData\Local\code\Arduino15\arduino-cli.yaml**
_board_manager:
    additional_urls: []
build_cache:
    compilations_before_purge: 10
    ttl: 720h0m0s
daemon:
    port: "50051"
directories:
    data: C:\Users\me\Documents\Arduino\libraries
    downloads: C:\Users\me\AppData\Local\Arduino15\staging
    user: C:\Users\me\Documents\Arduino
library:
    enable_unsafe_install: true
logging:
    file: ""
    format: text
    level: info
metrics:
    addr: :9090
    enabled: true
output:
    no_color: false
sketch:
    always_export_binaries: false
updater:
    enable_notification: true_

[image]

aklein24, 2024-01-15:
@kameo4242 sometimes being a stubborn donkey is the only way to solve some of these problems! Glad you got it working!

youneshenni, 2024-05-26:
One thing that might help is to set the name for the .ino file as the same name of the folder it's into, very weird but it worked for me

MatthewGumbert, 2024-07-23:
> I found I needed to actually open the Arduino IDE, verify some code, and install a library before the extension worked. I'm assuming this places compilation C/C++ files on the system that the [...]

yup, that worked thanks !