microsoft/vscode-arduino #1545: Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change.

koen1711, 2022-11-12:
So I have set:
"arduino.path" to: C:\Program Files\Arduino IDE
And restarted VScode and it still doesn't work.

What could be the issue?

rw377, 2022-12-01:
https://github.com/microsoft/vscode-arduino/issues/1548

hlovdal, 2022-12-26:
This issue can most likely be closed as a duplicate of #1548.

benmcmorran, 2023-02-16:
Thanks for the feedback. [Only the legacy IDE and Arduino CLI are currently officially supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

closed 2023-03-26 by github-actions[bot]