microsoft/vscode-arduino #1548: Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings.

Srijajum, 2022-11-21:
I have tried every path that i could find but it still didn't work. I downloaded the ide from the arduino site and not the microsoft store
[image]

gvautour, 2022-11-24:
I opened my VS Code as administrator and it found the correct arduino:path from then on.

Srijajum, 2022-11-26:
did not work

Srijajum, 2022-11-26:
didn't work

bbulakit, 2022-11-26:
Did you install Arduino IDE version 2.x.x ?
I had the same problem, but I fixed it by downloading Arduino CLI -> https://arduino.github.io/arduino-cli/0.29/
I installed it to default path (C:\Program Files\Arduino CLI)

Then, I went to Arduino Setting in Visual Studio Code
(File -> Preferences -> Setting -> search about "arduino")

[image]

Scroll down to bottom of the settings, I configured Arduino Path as "C:\\Program Files\\Arduino CLI"
and checked on Arduino Use Arduino Cli also

[image]

Let me know if it works.

luisgizirian, 2022-11-29:
I've followed @bbulakit latest instructions, but my scenario is a bit different: Arduino CLI is installed on a Raspberry Pi (with latest Raspian Lite 32 bit) to which I'm connected from my Mac through VSCode (Insiders) Remote SSH Extension.

Then, I've installed the Arduino Extension for the remote ssh (the Raspberry Pi), completed `arduino.path` with the right path and indicated I'm using the CLI.

I don't know if this scenario is contemplated, cause I'm still getting the subject error message as if I were using the IDE instead of the CLI.

luisgizirian, 2022-11-30:
I've been looking around the source code. Not sure how `internalSysLib` will "delegate" in my case

* Mac host running VSCode,  running VS Code Server on a Linux - Raspberry Pi Raspian OS - with Remote SSH extension.

https://github.com/microsoft/vscode-arduino/blob/main/src/common/platform.ts

What I can't discern is wether will think is running on a Mac, a Linux box, or in two parts (as actually is).

peregrinogris, 2022-12-03:
I had the same experience as @vindicatorr, figuring out what to put on each field. I'm on MacOS with the CLI installed through homebrew so it's in a slightly different path:

```
  "arduino.commandPath": "arduino-cli",
  "arduino.path": "/opt/homebrew/bin/",
  "arduino.useArduinoCli": true,
```

closed 2022-12-12 by Srijajum

vindicatorr, 2022-12-12:
I'll also go ahead and say that I don't think this issue ought to be closed until clarity is added into the code, otherwise others will keep encountering this issue.