microsoft/vscode-arduino #1550: Plugin did not work with Arduino IDE 2.x.x

milen-prg, 2022-11-24:
I have installed Arduino 2.0.2 on Windows 10 Pro 64bit.
The Arduino path in VSCode settings is correct:
C:/Users/mlap/AppData/Local/Programs/Arduino IDE/

But any Arduino command in VSCode fails with:
Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change.

rw377, 2022-12-01:
https://github.com/microsoft/vscode-arduino/issues/1548

milen-prg, 2022-12-05:
> 

There is no problem decision. I have installed Arduino IDE and many libraries in it, and I want all functions from VSCode, especially access to the libraries examples.

hlovdal, 2022-12-26:
This issue can most likely be closed as a duplicate of #1548.

K7MDL2, 2023-01-03:
See my comment in [#1557](https://github.com/microsoft/vscode-arduino/issues/1557)
Can use the 2.0 CLI if you enter the correct path to it.

benmcmorran, 2023-02-16:
Thanks for the feedback. [Only the legacy IDE is currently officially supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

closed 2023-02-16 by benmcmorran