microsoft/vscode-arduino #1557: Can't select board type

stylonurus, 2022-12-17:
Type: <b>Bug</b>

Just trying to get this IDE to work for me.
The message I get is "Cannot find Arduino IDE. Please specify the "arduino.path" in the User Settings. Requires a restart after change."    
Steps: After adding Arduino extention, I updated path to directory where Arduino IDE exe is located in order to select board type. (I verified it in the settings.json file)) 
I tried many times on two machines. Installed Arduino twice, once with zip, the second as a msi. Two different directory paths. I must be missing something. I really would love to use this IDE. Please help. My email is stylonurus@yahoo.com  

Extension version: 0.4.12
VS Code version: Code 1.74.1 (1ad8d514439d5077d2b0b7ee64d2ce82a9308e5a, 2022-12-14T10:30:51.966Z)
OS version: Windows_NT x64 10.0.19045
Modes:
Sandboxed: No

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-2600 CPU @ 3.40GHz (8 x 3392)|
|GPU Status|2d_canvas: unavailable_software<br>canvas_oop_rasterization: disabled_off<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: disabled_software<br>multiple_raster_threads: enabled_on<br>opengl: disabled_off<br>rasterization: disabled_software<br>raw_draw: disabled_off_ok<br>skia_renderer: enabled_on<br>video_decode: disabled_software<br>video_encode: disabled_software<br>vulkan: disabled_off<br>webgl: unavailable_software<br>webgl2: unavailable_software<br>webgpu: disabled_off|
|Load (avg)|undefined|
|Memory (System)|11.98GB (8.10GB free)|
|Process Argv|--crash-reporter-id b96425e7-70be-47e6-b971-ec873e7fbee2|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383:30185418
vspor879:30202332
vspor708:30202333
vspor363:30204092
vslsvsres303:30308271
pythonvspyl392:30443607
vserr242cf:30382550
pythontb:30283811
vsjup518:30340749
pythonptprofiler:30281270
vsdfh931cf:30280410
vshan820:30294714
vstes263:30335439
vscorecescf:30445987
pythondataviewer:30285071
vscod805:30301674
binariesv615:30325510
bridge0708:30335490
bridge0723:30353136
cmake_vspar411:30581797
vsaa593:30376534
pythonvs932:30410667
cppdebug:30492333
vsclangdf:30486550
c4g48928:30535728
dsvsc012cf:30540253
azure-dev_surveyone:30548225
pyindex848:30577860
nodejswelcome1cf:30587006
2e4cg342:30602488
gswce1:30612156
iaj6b796:30613358
f6dab269:30613381
fim-prod:30623723

```

</details>

hlovdal, 2022-12-26:
Which version of Arduino ide did you install, the [newer 2.x version](https://github.com/arduino/arduino-ide) or the [older 1.x version](https://github.com/arduino/Arduino)? According to the readme file

> The supported Arduino IDE versions are 1.6.x and up to, but not including, 2.0.0.

hlovdal, 2022-12-26:
This issue can most likely be closed as a duplicate of #1548.

stylonurus, 2022-12-26:
Yes, I am using 2.0 but I got the IDE to work adequately by using Visual Studio cide with Arduino extension but more importantly the PlatformIO extention.  I am now able to set breakpoints and look at variables in a watch window. 
If there is a better IDE, I would love to know about it. Thank you for getting back to me.
Best Regards, 
Rick

K7MDL2, 2023-01-03:
I have the extension working with 2.0 fairly well.  
For the IDE it was looking for arduino_debug.exe which appears to no longer exist in 2.0.  
Instead I enabled the CLI and entered the full path to it, it ships with the 2.0 IDE.  Just have to know where to find it.
It is located in your installation folder as below.
...\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\

benmcmorran, 2023-02-16:
Thanks for the feedback. [Only the legacy IDE is currently officially supported](https://github.com/microsoft/vscode-arduino/blob/374c78f8ed06c1e88df480b687c24c06b728e13b/README.md?plain=1#L39). Additionally, **we plan to drop support for the legacy IDE entirely in the next release of this extension** and recommend that all users move to Arduino CLI as soon as possible. The recent `v0.5.0` release of the extension includes a bundled copy of Arduino CLI that should ease this migration.

closed 2023-03-26 by github-actions[bot]