microsoft/vscode-arduino #1567: Language server error related to generated compilerPath in c_cpp_properties.json for esp8266/Arduino

milanbx, 2023-01-15:
## Problem
After upgrading *ESP8266/Arduino core* to *v3.1.1*, the extension regenerated th `c_cpp_properties.json` file with a `compilerPath` attribute having value causing *C/C++ Configuration error*:
`Unable to resolve configuration with compilerPath: "@D:\Portable\Arduino\portable\packages\esp8266\hardware\esp8266\3.1.1\tools\warnings\default-g++"`
On top of this line from `C/C++ Configuration Warning` output, the Code UI shows 5 instances of `cpptools` output, all having same content (see also snapshot below):
`[Error - 18:20:27] Connection to server got closed. Server will not be restarted.`
[image]
Then, the intellisense stops warking at all, co cross reference can be followed (Ctrl+left mouse click) to see details of function or variable declaration etc.

## Environment
### Code
Version: 1.74.3
Commit: 97dec172d3256f8ca4bfb2143f3f76b503ca0534
Date: 2023-01-09T16:59:02.252Z
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: No

### Extensions
Arduino: 0.4.12
C/C++ Extension Pack: 1.3.0

## Steps to reproduce
Just run VS Code, open any sketch with Arduino code (*.ino), even empty one. Popup error start to come up in the lower right corner - see snapshot:
[image]

## Code snippet
Every, does not matter, even this simple blinking LED example:
``` cpp
byte go = LOW;
void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, go);
}
void loop() {
    delay(500);
    go = 1 - go;
    digitalWrite(LED_BUILTIN, go);
}
```

## Dev Tools Console Log
see attached file please
[vscode-app-1673804966418.log](https://github.com/microsoft/vscode-arduino/files/10420833/vscode-app-1673804966418.log)

milanbx, 2023-01-19:
I have used vanilla Code portable installation v1.74.3 with only Arduino extension, which requires C/C++ extension v1.13.9, without ESP8266 or any other aditional board, and empty sketch configured for Arduino Uno (on Windows 11), and the bug did not appear any more. It meand, the bug has to do something with ESP8266 Arduino core.

chibidev, 2023-01-27:
ESP8266 Arduino core changed how it handles compiler warnings with esp8266/Arduino#8495 (released in probably 3.1.0), which resulted in vscode's compiler output parser to pick up the wrong path for the compiler. esp8266/Arduino#8829 fixes that, hopefully we get a release soon.

Until then, a temporary fix might be to rename each file ending in -g++ and -gcc to something else (e.g. extra-g++ -> extra-cpp and alike). Don't forget to change `platform.txt` as well.

closed 2023-01-30 by milanbx